realsenseai/librealsense #8978: IR Stream in the dark room (white laser dots cleaning)

DuyguSerbes, 2021-05-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | Open RealSense Viewer | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-72-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |   l2.0                          | 
| Language                            |  Python                           | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello all,

I want to get infrared stream in the dark environment conditions. In order to do that I increase laser power but it creates white laser dot on the image. How can I get rid of this kind of issues? When I drop laser power to 0 or 1, the white laser dots dont be visible but rest of image become black. 

Do you have any solution to get clean IR stream in the dark room?

MartyG-RealSense, 2021-05-06:
Hi @DuyguSerbes  Dot noise on the image is a phenemon known as **laser speckle**.  A way to help to avoid laser speckle is to use an external **LED** pattern projector instead of a laser-based one like the one built into the camera.  An example of an external LED projector used with a RealSense 400 Series camera in a dark room is shown in the image below.

[image]

Intel have published a white-paper document that provides guidance about external pattern projectors.

https://dev.intelrealsense.com/docs/projectors

https://dev.intelrealsense.com/docs/projectors#section-7-led-projectors

DuyguSerbes, 2021-05-06:
Hello @MartyG-RealSense even if I didn't set a maximum, the bright white dots covers the image. I don't have external illuminator. Is there a any solution for this problem? I want to use infrared images for computer vision so I need clean images.

MartyG-RealSense, 2021-05-06:
You could enable the **Emitter On-Off** option so that the emitter rapidly alternates between on and off on a per-frame basis, and create a script that only captures frames where the emitter status is 'off'.  

You can test whether this solution may work for you in the RealSense Viewer.  Enable the **Emitter On-Off** option and on the infrared stream, tap the space bar on the keyboard quickly to pause and unpause the stream.  When you manage to pause at a frame where the dot pattern is inactive, check whether the infrared image is still visible when the dot pattern is off.

[image]

This pause-unpause technique is demonstrated in the images below.

[image]

[image]

DuyguSerbes, 2021-05-06:
I believe It would work. Is there any available script written in C/C++ or Python to manage Emitter on/off option. 

I try to use frames directly in my code. So I should take the frames directly from camera without using realsense-viewer.

MartyG-RealSense, 2021-05-06:
There is a Python script that features enabling of the emitter on-off feature in the opening message of the link below.

https://github.com/IntelRealSense/librealsense/issues/6202

[image]

The C++ equivalent would be something like:

depth_sensor.get_option(RS2_OPTION_EMITTER_ON_OFF, 1);

closed 2021-05-18 by MartyG-RealSense