realsenseai/librealsense #8981: While recording depth align to color using align-adv sample code, depth image is zoomed out compared to color image.

adharma1, 2021-05-07:
<html>
<body>

Required Info |  
-- | --
Camera Model | D435i
Firmware Version | 5.12.11
Operating System & Version | Linux (Ubuntu 18)
Kernel Version (Linux Only) | 5.4.70
Platform | VM
SDK Version | 2.42.0
Language | C++
Segment | {Robot/Smartphone/VR/AR/others }

</body>
</html>
### Issue Description
I am doing a recording of depth alignment (to color) and storing it in the rosbag, when I look at the aligned depth and color image I see the depth image is zoomed out compared to the color image, I use the standard align-adv sample code to do the recording.

adharma1, 2021-05-07:
[image]
[image]

MartyG-RealSense, 2021-05-07:
Hi @adharma1   Bag files do not store aligned images.  They store the streams separately.  When the bag file is loaded into memory, an align operation can then be performed on the data in real-time.

adharma1, 2021-05-07:
@MartyG-RealSense Do have sample code using which I can do this?

adharma1, 2021-05-07:
Thanks a lot for quick response!

MartyG-RealSense, 2021-05-07:
Intel have a Jupyer Notebook tutorial for Python called **distance_to_object** that demonstrates aligning depth and color from a bag.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

MartyG-RealSense, 2021-05-07:
I see you are using C++.  You could use the SDK's **rs-align** sample program for live-camera alignment and adapt it to use a recorded bag file as the data source instead.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

The first paragraph of the link below explains how to adapt the code of a live-camera example program to read a bag instead.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

MartyG-RealSense, 2021-05-10:
You can certainly generate aligned topics with ROS and the RealSense ROS wrapper by setting a condition called **align_depth** to True.

**align_depth: If set to true, will publish additional topics for the "aligned depth to color" image.: /camera/aligned_depth_to_color/image_raw, /camera/aligned_depth_to_color/camera_info.  The pointcloud, if enabled, will be built based on the aligned_depth_to_color image.**

A bag file can be recorded in ROS instead of the RealSense SDK with **rosbag record**, and using the instruction **rosbag record -all** will record all currently  published topics to a bag.  Having said that, I am not aware of past cases where recording aligned topics to bag has been demonstrated.

http://wiki.ros.org/rosbag/Commandline#rosbag_record
http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data

adharma1, 2021-05-11:
@MartyG-RealSense Thanks a lot for the info! I wanted to integrate this using realsense SDK, I have seen the rosbag method, but we don't want to invoke binaries during execution. Let me know if this can be achieved using realsense SDK.

adharma1, 2021-05-11:
Can the following be achieved?
Let's say I save rosbag using realsense SDK as "rosbag1.bag", Can I use some python/c++ code to read "rosbag1.bag" and do alignment and write it into "rosbag2.bag"?

MartyG-RealSense, 2021-05-11:
I researched your problem carefully today.  The most common scenario when handling bag data with the RealSense SDK instead of ROS is to load individual streams from the bag into the computer's memory and then align them in real-time.  Performing alignment and then re-saving the bag would not solve the problem unfortunately, as the aligned data would still be saved into the bag by the SDK as separate streams.

An alternate approach would be to use OpenCV in your application to generate an aligned **cv** image and then save the contents of that OpenCV image to a video file such as mp4.   An example of such an operation with OpenCV and C++ or Python is provided in the link below.

https://learnopencv.com/read-write-and-display-a-video-using-opencv-cpp-python/

My research did not find a way to afterwards convert the video into a bag file from within the same application though.

MartyG-RealSense, 2021-05-11:
Yes, as separate depth and color topics.  The SDK's bag structure is based on the ROS rosbag standard but has some differences that can cause incompatibility when trying to read an SDK-created bag in ROS or a ROS-created bag in the SDK.

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

ROS bags are typically recorded using ROS' own **rosbag record** tool instead of a RealSense-specific tool.

MartyG-RealSense, 2021-05-11:
I do not know of a way to achieve what you described above.  I researched the question but could find nothing, unfortunately.

MartyG-RealSense, 2021-05-11:
I will seek advice from Intel.  Thanks very much for your patience.

MartyG-RealSense, 2021-05-12:
Hi @adharma1  After consultation with the Intel RealSense team, the response was a confirmation of my earlier advice.  
The RealSense rosbag is a recording of raw data from the sensors.  Alignment data is not saved into the bag and alignment and post-processing are performed in real-time after the bag is loaded.

The advice also stated the following points:

-  There is no tool to convert a RealSense rosbag to a ROS rosbag.

-  The **rs-convert** SDK tool can extract frames and metadatas from a RealSense rosbag, but it may not be of help because they still perform the align frame and convert them back to rosbag. 

-  The most straightforward way to capture aligned images would be to use RealSense ROS to enable and publish the aligned_depth topics and record the topics using the generic ROS recorder.

https://github.com/IntelRealSense/realsense-ros#aligned-depth-frames

adharma1, 2021-05-13:
@MartyG-RealSense would you be aware of implementation which integrates the generic ROS with c++ for real sense recording?

MartyG-RealSense, 2021-05-13:
I do not know of a ROS rosbag recording solution within the SDK in C++, though a RealSense user did write a rosbag parser for **reading** rosbags.

https://github.com/IntelRealSense/librealsense/issues/2215

closed 2021-05-25 by MartyG-RealSense