realsenseai/librealsense #8988: D435i how to get left IR image and right IR image

lndawn, 2021-05-10:
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
when I set the config as described above,the function pipe.start(cfg) will report error. this error occur when I use the latest SDK and firmware.

MartyG-RealSense, 2021-05-10:
Hi @lndawn  Is your camera on a USB 2 connection, please?  If it is attached to a USB 3 port then there is also the possibility that it is being mis-detected as a USB 2 connection.

If the camera is being detected as being on USB 2 then Infrared 2 would not be not available as it requires a USB 3 connection to be accessed.  In that situation, you may receive a **Couldn't resolve requests** error, signifying that the SDK is unable to provide the requested configuration because it is not currently supported by the camera (for example, due to being in USB 2 mode) or the request is not understood (for example, due to a typing error, which is not the case here).

Also, 30 FPS would not be supported on a USB 2 connection, as the camera has more limited capabilities than when using USB 3 and so would be limited to 6 FPS at 1280x720 resolution.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

You can check whether the camera is being detected as USB 2 or 3 by launching the RealSense Viewer program and looking at the number beside the camera name at the top of the options side-panel.  If the number is **2.1** then the camera is being recognized as being in USB 2 mode.

[image]

lndawn, 2021-05-10:
thanks a lot your detailed answer, when I change the connected line the question is solved

closed 2021-05-15 by MartyG-RealSense