realsenseai/librealsense #8994: Realsense D455 not connecting to Raspberry Pi 4

cryptoscientia, 2021-05-10:
Hello everyone,  

I have been attempting to hook up a Realsense D455 to a raspberry Pi 4 for about a week now.  It seems that everything I do it can not be found.

First info on the raspberry pi, running:
`cat /etc/os-release`

I get:
`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"`

For the kernel:
`uname -a`

I get:
`Linux raspberrypi 5.10.35-v7l+ #1417 SMP Fri May 7 17:44:06 BST 2021 armv7l GNU/Linux`

The librealsense version is: 2.33.1

Please let me know if any other versions or checks needed.  
---------------------------------------------------------------------------------------------------------------
Now for the issue, when running any program, be it RTAB-Map, Realsense-viewer (or via terminal line), or the python SDK wrapper named Pyrealsense2; the result is the same it can not find the D455.  

I do have a realsense D435 and I find it with all the above stated library's or GUI's and it works just fine.  I will also state I have been able to get all of this working on a windows machine with the D455 just fine.  

But, for example, running the below command with the D455:
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

I get:
`[ WARN] [1620682361.605577449]: No RealSense devices were found!`

Following what I could find online searches I also found:
[https://github.com/IntelRealSense/librealsense/issues/7390](url)
[https://github.com/IntelRealSense/librealsense/issues/7312](url)

I had went through in detail attempting to follow all advise given within the above two links.  But still am having this issue.  I have already started from scratch twice.  I turn to here because I can not seem to get this working.  It appears people out there also have this issue still too.  Hopefully this post will help solve this once and for all.  Thank you.

MartyG-RealSense, 2021-05-10:
Hi @cryptoscientia   The D455 is a much newer camera model than the D435, so the SDK version 2.33.1 that you are using did not yet have D455 support.  SDK version **2.38.1** or newer should be used with D455.

cryptoscientia, 2021-05-11:
Yes, still no change.  I even tried the black USB 2 ports.

MartyG-RealSense, 2021-05-11:
Could you attach the D435's confirmed-working USB cable to the D455 please to eliminate the possibility that the D455 has a bad USB cable?

cryptoscientia, 2021-05-12:
Confirmed, the cable is not the issue.  It works just fine with Windows but not on the Raspberry Pi.

MartyG-RealSense, 2021-05-12:
Let's go back to the beginning to seek some fresh insights.

What process did you use to install the librealsense SDK and the RealSense ROS wrapper on Raspbian Buster please?

For the Pi 4, the most popular installation guides for Raspbian installation are:

**Acrobotic - Librealsense and Python**
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

**Eleccelerator - Librealsense, Python and ROS**
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

However, the fact that you are able to use the D435 successfully on your Pi 4 suggests that the install method that you have been using is not at fault.

It is a surprise to me if you have managed to use D435 on the Pi with Raspian, as the only successful uses of a 400 Series camera in ROS on Pi 4 that I know of have been ones that used Ubuntu Server as the OS instead of Raspbian.

cryptoscientia, 2021-05-12:
Thanks for your patience and responses here.  I can wipe the memory once more and try from the beginning again (not an issue for me at all to do).

Let me step back and let you know what I'm try to do.  I'm looking to make a handheld 3D scanner using the D455.  

See:
[https://eleccelerator.com/pi-handheld-3d-scanner/](url)

Here they us a raspberry pi with a D435 to make the hand held scanner.  This is my goal.

As for the above link, I have been successful with the D435, but run into the issues that have been discussed here and is why I'm reaching out for help.

What I will do is wipe my pi and start again with a fresh implementation.  I'll look into the links you provided as the installation guide, in fact I have run across the Eleccelerator guide before.

Let me start from scratch and hopefully install the correct librealsense version right of the bat. I'll let you know what it says after that is done.

MartyG-RealSense, 2021-05-12:
Thanks very much for your patience too during this case.  It's very appreciated.  Good luck with the fresh installation!

sam598, 2021-05-13:
> It is a surprise to me if you have managed to use D435 on the Pi with Raspian, as the only successful uses of a 400 Series camera in ROS on Pi 4 that I know of have been ones that used Ubuntu Server as the OS instead of Raspbian.

FWIW I use the D435 and D455 with Raspbian/Raspberry Pi OS on a Raspberry Pi 4 all the time. Starting with the Pi4 the OS has a built in GL driver which eliminates the need for an alternate OS.

Very recently I tried building the SDK using "Raspberry Pi OS Lite" to try and solve some performance issues. Even though the SDK built without any errors, it would not recognize any D400 camera even though it appeared under lsusb.

I added all of the `sudo apt install` dependancies listed in:
> Acrobotic - Librealsense and Python
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435
Eleccelerator - Librealsense, Python and ROS
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

rebuilt the SDK, and now it works as expected. Unfortunately I do not know which specific dependency was the culprit.

cryptoscientia, 2021-05-13:
Ok so I have reloaded as carefully as I am able an ensured that I had the latest packages as far as I could tell. 

I used the:
[https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense](url)

Here is what I found.  The D435I worked no issues.  

The D455 did work, however something was off.  It did show the depth and the camera, it seemed a little lagged though.

An error pops up in a dialog box and says:
Oops, something went wrong!
Realsense error calling:
Invalid Value in rs2_get_option(options:0xae712618, option:Emitter On Off): hwmon command 0x7d failed.
Error type: No data to return (-21).

I did not quite know how to put the error so I just typed it out above.  

If you select ok, the camera still works, it just is lagged and glitchy. 

Any thoughts?

MartyG-RealSense, 2021-05-14:
Thanks very much @sam598 for your advice to @cryptoscientia 

Hi @cryptoscientia   Does the performance of the D455 in ROS improve if you add **initial_reset:=true** onto the end of your roslaunch instruction to reset the camera at launch?  For example:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true**

cryptoscientia, 2021-05-14:
Well the command does not work for the D455, but does for the D435. 

I installed RTAB-Map and now I'm back to the error I had before:
`[ WARN] [1620682361.605577449]: No RealSense devices were found!`

MartyG-RealSense, 2021-05-14:
The eleccelerator instructions are somewhat outdated now in regard to their reliance on ROS Kinetic, especially as ROS Kinetic has just reached end-of-life.  

https://discourse.ros.org/t/preparing-for-final-kinetic-sync/19805

If I were using the guide myself then I would be inclined to adapt the Kinetic references to use the more modern ROS Melodic or Noetic instead.

cryptoscientia, 2021-05-17:
Ok sounds good, I will try that today and let you know how it goes.  Thanks!

MartyG-RealSense, 2021-05-23:
Hi @cryptoscientia  Do you have an update about this case that you can provide, please?   Thanks!

cryptoscientia, 2021-05-24:
Hello MartyG, I was planning on getting to it tomorrow, will provide an update when I'm through.  Thanks!

MartyG-RealSense, 2021-05-25:
Thanks very much @cryptoscientia - good luck!

cryptoscientia, 2021-05-28:
Hello MartyG-RealSense,

I started with the basic Rastaban package for a raspberry pi, totally fresh, nothing but what it comes with.

So I will list all steps I needed to install the Realsense (or at least what I did to get a proper install):
`sudo apt-get update
sudo apt-get upgrade

sudo nano /etc/dphys-swapfile
     -> CONF_SWAPSIZE=2048 (change within file)

sudo /etc/init.d/dphys-swapfile restart swapon -s

sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbgsym libdrm-dev libdrm-exynos1 libdrm-exynos1-dbgsym libdrm-freedreno1 libdrm-freedreno1-dbgsym libdrm-nouveau2 libdrm-nouveau2-dbgsym libdrm-omap1 libdrm-omap1-dbgsym libdrm-radeon1 libdrm-radeon1-dbgsym libdrm-tegra0 libdrm-tegra0-dbgsym libdrm2 libdrm2-dbgsym

sudo apt-get install -y libglu1-mesa libglu1-mesa-dev glusterfs-common libglu1-mesa libglu1-mesa-dev libglui-dev libglui2c2

sudo apt-get install -y libglu1-mesa libglu1-mesa-dev mesa-utils mesa-utils-extra xorg-dev libgtk-3-dev libusb-1.0-0-dev

sudo apt-get install build-essential libtool autoconf automake

cd ..
git clone git://git.openssl.org/openssl.git --depth 1
cd openssl
./config
make
make test
sudo make install

git clone --depth=1 -b v3.5.1 https://github.com/google/protobuf.git
cd protobuf
./autogen.sh
./configure
make -j1
sudo make install

cd ~
(downloaded Cmake-3.20.3)
tar -zxvf cmake-3.20.3/
cd cmake-3.20.3/
./configure --prefix=/home/pi/cmake-3.20.3
make -j1
sudo make install
export PATH=/home/pi/cmake-3.20.3/bin:$PATH

sudo apt-get install libxcb-randr0-dev libxrandr-dev
sudo apt-get install libxcb-xinerama0-dev libxinerama-dev libxcursor-dev
sudo apt-get install libxcb-cursor-dev libxkbcommon-dev xutils-dev
sudo apt-get install xutils-dev libpthread-stubs0-dev libpciaccess-dev
sudo apt-get install libffi-dev x11proto-xext-dev libxcb1-dev libxcb-*dev
sudo apt-get install bison flex libssl-dev libgnutls28-dev x11proto-dri2-dev
sudo apt-get install x11proto-dri3-dev libx11-dev libxcb-glx0-dev
sudo apt-get install libx11-xcb-dev libxext-dev libxdamage-dev libxfixes-dev
sudo apt-get install libva-dev x11proto-randr-dev x11proto-present-dev
sudo apt-get install libclc-dev libelf-dev git build-essential mesa-utils
sudo apt-get install libvulkan-dev ninja-build libvulkan1 python-mako
sudo apt-get install libdrm-dev libxshmfence-dev libxxf86vm-dev libunwind-dev
sudo apt-get install valgrind libzstd-dev vulkan-tools vulkan-utils

sudo apt-get update
sudo apt-get upgrade

git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo su
     -> sudo udevadm control --reload-rules && udevadm trigger
     -> exit
mkdir build
cd build/
cmake ..
make -j4
sudo make install

realsense-viewer
`

At this point the realsense-viewer will turn on and will show the GUI for the realsense.  BUT the D435i or the D455 do not show up or even get detected for some reason.  I may have missed something?  Please let me know if I should try adding anything.
Thanks

MartyG-RealSense, 2021-05-28:
Thanks very much for the very detailed response.  For Pi 4 Raspbian installations of RealSense, I usually use the **Acrobotic** guide that is popular with RealSense users as my main reference.  

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Your method listed above has a lot of the same elements as the Acrobotic guide but is also missing some of Acrobotic's steps.  

For example, its CMake instruction tells CMake to build librealsense, **plus** build the examples and tools, compile the SDK in 'Release' mode with optimizations, and use the DFORCE_LIBUVC mode to buld librealsense without dependency on the kernel.

**cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true**

The Acrobotic guide also installs OpenGL, which is required by the RealSense Viewer, and enables OpenGL on the Pi.

**sudo apt-get install python-opengl
sudo -H pip3 install pyopengl
sudo -H pip3 install pyopengl_accelerate==3.1.3rc1**

**sudo raspi-config
"7. Advanced Options" – "A8 GL Driver" – "G2 GL (Fake KMS)"**

cryptoscientia, 2021-05-28:
Success! (I hope!) Half of the problem is solved, I have both cameras, the D35i and D455 work.  I do get a bunch of errors but nothing seemed to break...

 ```
28/05 13:43:58,496 WARNING [2858402880] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 28/05 13:43:58,646 WARNING [2858402880] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 28/05 13:43:58,696 WARNING [2858402880] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No such device, number: 13
 28/05 13:43:58,708 WARNING [2641359936] (ds5-device.cpp:156) usb device disconnected
 28/05 13:44:27,414 ERROR [2477700160] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/05 13:44:27,414 ERROR [2641359936] (sensor.cpp:526) acquire_power failed: failed to set power state
 28/05 13:44:27,415 WARNING [2641359936] (rs.cpp:306) null pointer passed for argument "device"
 28/05 13:44:27,583 ERROR [2630874176] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/05 13:44:27,583 ERROR [2641359936] (sensor.cpp:526) acquire_power failed: failed to set power state
 28/05 13:44:27,584 WARNING [2641359936] (rs.cpp:306) null pointer passed for argument "device"
 28/05 13:44:41,329 WARNING [2595198016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 13:44:41,379 WARNING [2595198016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 13:44:41,430 WARNING [2595198016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 13:44:49,620 WARNING [3069944976] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 28/05 13:44:55,478 WARNING [2687956032] (sensor.cpp:342) Frame received with streaming inactive,Color0, Arrived,0.000000 1622234695467.595947

```

Above is what is printed in the terminal when running realsense-viewer.

Now I'm going to try to get the Rtab-Map installed.
I'll update you when I get that.
Thanks

MartyG-RealSense, 2021-05-28:
You're very welcome @cryptoscientia - great to hear that you've made significant progress!

MartyG-RealSense, 2021-06-09:
Hi @cryptoscientia  Do you have an update that you can provide about your goal of getting RTABMAP installed please?  Thanks!

cryptoscientia, 2021-06-09:
Getting the RTABMAP installed was the easy part.  I built it from source.  

The trick was to ensure you have the latest RTABMAP and then to choose the Realsense driver for the camera.  The D455 worked.  

However it was laggy, I set the image decimation to 2 and limited the ranges.  This provided a slightly faster capture rate for the Raspberry Pi.  I also made it a portable device.  Although scanning was slow, it worked.  

I also tried it with various Micro SD cards, including the highest write rate I could find at 300 Mb/sec.  This also improved only slightly the speed at which it ran.   Lastly, I found the best heat sink I could find made of a solid aluminium chunk.  This too also helped the lag. 

Ultimately, the Raspberry Pi will work, however you have to put up with the lag.  

I moved to the Xavier Nvidia AGX computer running Ubuntu 20 and installed the Realsense and RTABMAP fairly similar to how it was done for the raspberry pi, with much more improvement on speed, however less portable. 

It was a fun project for sure.

MartyG-RealSense, 2021-06-09:
Thanks so much @cryptoscientia for sharing your project experiences in detail for the benefit of others reading this case!  I'm pleased to hear that it went smoothly for you.  :)

MartyG-RealSense, 2021-06-16:
Hi @cryptoscientia  Do you require further assistance with this case, please?  Thanks!

physics-fm, 2021-06-17:
I'm doing well, I actually tweaked the parameters on the RTABMAP, such as the decimation == 2 and setting a min/max for the limits, it helps speed it up and produce a faster scan on the raspberry pi.  

As for the Xavier Nvidia AGX, it fully works and is super fast with almost no loss.  

My take is this:
Raspberry Pi - Cheap, slow, easily portable, and decent scan.  Works with D455 and D435i, however I find it works better with the D435i.
Xavier Nvidia AGX - Expensive, fast, not as portable (still possible but with right equipment/power source), and incredible scans.  Works with D455 and D435i, but the D455 works much better. 

For anyone looking to replicate the project, I would say it is absolutely possible but I would also say you could run into the same issues installation wise as stated here.  Just follow the instruction set given above and read the comments carefully and it should work.  Note: this was done June 2021, so keep in mind that future releases of OS and updating software may change the solution above. But that being said, I started with a 'from scratch' new raspberry pi and was able to make it work on two different Raspberry Pi's.  Good luck!

I good if you want to end this, no further assistance needed.  Thank you for all your help!

MartyG-RealSense, 2021-06-17:
Thanks very much @physics-fm for your very detailed feedback about the method that worked for you!

 Are you an alternate account of @cryptoscientia  ?   :)

physics-fm, 2021-06-17:
Lol, I did not realize that it was under @physics-fm...  But that one is a project I'm helping out with.

@cryptoscientia is my personal one.  :-)

MartyG-RealSense, 2021-06-17:
Thanks very much for the confirmation - if you are satisfied with the outcome, you are welcome to close this case with the **Close Issue** button underneath the comment box next time you are logged in as @cryptoscientia  :)

closed 2021-06-17 by cryptoscientia

mgagvani, 2021-08-10:
Hello, I am encountering an issue similar to one in this thread. 

I compiled librealsense 2.47 from source , following [this](https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435) guide. At this point realsense-viewer worked fine. However, I had forgotten to compile the python bindings so I deleted all realsense files from /usr/local/bin. /usr/local/lib, and /usr/local/include and rebuilt with the python bindings. However, when I run realsense-viewer now, I am getting the error below. 

Same issue is found here: https://github.com/IntelRealSense/librealsense/issues/8994#issuecomment-838787661

Here is a sample of the error log:

`21:13:20 [Info] /media/pi/ESD-USB1/pyrealsense2/librealsense/src/linux/backend-v4l2.cpp:565 - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:3/device/../../../../../../../../../..`

I have already tried these things:

- Tried both L515 and D455, neither worked. 
- Used the USB-C cable that **came with the cameras.**
- Plugged it in quickly and firmly, into both USB 3(blue) ports.

Complete log: [log.txt](https://github.com/IntelRealSense/librealsense/files/6958309/log.txt)

Thanks in advance.

MartyG-RealSense, 2021-08-10:
Hi @mgagvani   In case number https://github.com/IntelRealSense/librealsense/issues/8195#issuecomment-763858898 where this busnum/regnum error also occurred with the Viewer, the RealSrnse user in that case resolved the error by building librealsense from source code with the CMake build term **-FORCE_RSUSB_BACKEND=true**

Using the RSUSB method of installing librealsense requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

mgagvani, 2021-08-10:
Thanks for the help @MartyG-RealSense. Now both realsense-viewer and the python bindings work perfectly!

MartyG-RealSense, 2021-08-10:
It's great to hear that you had a successful outcome, @mgagvani   :)