realsenseai/librealsense #8996: pyrealsense2.so cannot open shared object file : No such file or directory

b2992872, 2021-05-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |   legacy / 2.0                      | 
| Language                            |  python                       | 
| Segment			|                         | 

### Issue Description

I put pyrealsense2.so and librealsense2.so together, but it cannot be executed successfully
[image]
[image]

MartyG-RealSense, 2021-05-11:
Hi @b2992872   Are you using Python 2.7 or a Python 3 version, please?  The Python 3 version of the **pyrealsense2.so** file is likely to have a more complex filename that is generated based on the configuration of your computer.  

For example, as mentioned by the Python wrapper's Ubuntu installation instructions, a Python 3.5 installation may have a pyrealsense2.so filename such as **pyrealsense2.cpython-35m-arm-linux-gnueabihf.so** and that is the file that should be placed next to your script alongside librealsense2.so.

If you are using Python 3, I would suggest performing a file search for the filename **pyrealsense2.so** to locate where the exotically named Python 3 version of the file is located, as searching for the simple version of the filename should hopefully include the more complex versions of the filename in the search results too.

If you continue to have difficulties, there is an alternative installation method for building librealsense and the Python wrapper together from source code with CMake on Jetson devices that has worked consistently.

https://github.com/IntelRealSense/librealsense/issues/7905#issuecomment-737544294

b2992872, 2021-05-12:
I checked the python version in two ways, I am not sure which one I am using?
[image]

MartyG-RealSense, 2021-05-12:
My understanding is that if Python 3 is installed then the RealSense Python wrapper tries to use that instead of Python 2.7, as described in point 2 of the official instructions for building the wrapper from source code.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

[image]

b2992872, 2021-05-16:
Then what is the reason why python3 has such an error?( cannot open shared object file : No such file or directory)

MartyG-RealSense, 2021-05-16:
May I confirm that you have placed the files librealsense2.so and your exotically-named pyrealsense2.so file (e.g **pyrealsense2.cpython-35m-arm-linux-gnueabihf.so**) into the same folder as the one that your **realsense_depth.py** script is in?  That folder seems to be **home/nvidia/librealsense/build**

b2992872, 2021-05-16:
Yes like in the picture
[image]

MartyG-RealSense, 2021-05-16:
As mentioned earlier, if you are using Python 3 then when you built the Python wrapper there should be a complex-named version of the **pyrealsense2.so** file such as pyrealsense2.**cpython-35m-arm-linux-gnueabihf**.so.  Are you able to find a file like this on your computer and place it in the above folder?

b2992872, 2021-05-16:
I'm sorry I didn't see a similar file

MartyG-RealSense, 2021-05-16:
Okay, so let's assume the possibility that the exotic version of the filename does not get generated on Jetson installations and that pyrealsense2.so is the filename generated when the wrapper is built.

As the image in your first comment shows that you are already in the same **home/nvidia/librealsense/build** directory as the files when you launch the files, is there any difference to the outcome if you simply shorten the launch command to: **python 'realsense_depth.py'**

Does it also not work if you specify to launch it as a Python3 application with **python3 'realsense_depth.py'**

b2992872, 2021-05-17:
I tried to use python or python3 to execute, but the results are the same, maybe I should rebuild from the source

MartyG-RealSense, 2021-05-17:
For building the Python wrapper on Jetson, I usually recommend building the librealsense SDK and Python together at the same time using instructions in the link below, which have worked consistently for Jetson boards.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

b2992872, 2021-05-18:
Ok i will try, thanks!

MartyG-RealSense, 2021-05-24:
Hello, do you have an update about your case that you can provide, please?  Thanks!

closed 2021-05-30 by MartyG-RealSense