realsenseai/librealsense #8997: Experiencing runaway Ram usage while camera is on

mccormickjason, 2021-05-11:
| | |
|---|---|
|**librealsense**|2.44.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|801212070143|
|**Firmware Version**|05.12.12.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|751313020612|
|**Firmware Update Id**|751313020612|

Please provide a description of the problem

mccormickjason, 2021-05-11:
While the stereo process is streaming, but not recording, the RAM usage on PC the camera is hooked up to experiences runaway RAM usage

MartyG-RealSense, 2021-05-11:
Hi @mccormickjason   Are you using the RealSense Viewer or an application that you have created yourself, please?

If it is an application that you have created yourself, what does it do (for example, does it generate a point cloud).

mccormickjason, 2021-05-11:
It is the D435 stereo camera that was causing the RAM allocation issue. NOT the T265.

MartyG-RealSense, 2021-05-11:
Okay, thank you.  I can continue handling the case if T265 is not involved in the problem and it is not mentioned in the case from this point onwards.

As a starting point in investigating your case, could you try disabling the two **GLSL** options in the Viewer's **Settings** interface using the instructions in the link below please and see whether that makes a positive difference to RAM usage.

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

mccormickjason, 2021-05-13:
Thanks. Have not yet had the opportunity to attempt this fix. If possible, please leave this ticket open for approximately one week.

MartyG-RealSense, 2021-05-13:
Thanks very much for the update.  Yes, it is no problem to leave this case open for a further time period.  Good luck!

closed 2021-05-30 by MartyG-RealSense