realsenseai/librealsense #8999: How to change the resolution when saving a .ply point cloud file from L500/D400 using C++ Realsense2 library

cheer-Yuan, 2021-05-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 ( or D435) | 
| Firmware Version                   | 0003aa322188 | 
| Operating System & Version | Ubuntu 20 | 
| Kernel Version (Linux Only)    |  5.4.0.70 | 
| Platform                                 | PC |
| SDK Version                            | Latest | 
| Language                            |  C++  | 
| Segment			|   | 

Hello, 

I am searching for a method ( if it exists) to lower the resolution of .ply file generated by using the function 
```
void points::export_to_ply(const std::string& fname, const frame_holder& texture)
```

And my usage is
```
rs2::frameset frames = pipe.wait_for_frames();
auto depth = frames.get_depth_frame();
auto color = frames.get_color_frame();
pc.map_to(color);
rs2::points points = pc.calculate(depth);
points.export_to_ply("mypointcloud.ply", color);
```

I discovered that the .ply file generated in this way contains about 280000 vertices so I think that the resolution is 640*480. I am wondering that if it is possible to lower the resolution to 320*240 in order to make the processus of saving the .ply file faster.

RealSenseSupport, 2021-05-12:
@cheer-Yuan Please refer to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/latency-tool/rs-latency-tool.cpp#L13-L22 to configure the resolution you want. Thanks!

cheer-Yuan, 2021-05-12:
> @cheer-Yuan Please refer to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/latency-tool/rs-latency-tool.cpp#L13-L22 to configure the resolution you want. Thanks!

This solved my question. 

For information, I used \:
```
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 320, 240, RS2_FORMAT_ANY, 0);
pipe.start(cfg);
```
The size of the .ply file comes to about 2.8Mb from 10Mb, so I considered that the code works.

cheer-Yuan, 2021-05-12:
> @cheer-Yuan Please refer to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/latency-tool/rs-latency-tool.cpp#L13-L22 to configure the resolution you want. Thanks!

big thanks

closed 2021-05-12 by cheer-Yuan