realsenseai/librealsense #9000: [question] question about stream index of infrared streams

UnaNancyOwen, 2021-05-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
This is a question about stream index of infrared streams.
I noticed this strange specification in issue response of my tool.

RealSense D400 series (RealSense D415, RealSense D435, RealSense D435i) has two Infrared cameras.
Therefore, it has two Infrared streams, and each stream has stream index.
RealSense L500 series (RealSense L515) has one Infrared camera.
Therefore, it has one Infrared stream, and that stream has stream index.

So far, this is a confirmation of the specification. It is ok.
My question starts here.

RealSense D400 series's Infrared streams has stream index of 1 or 2. **It starts from 1.**
RealSense L500 series's Infrared stream has stream index of 0. **It starts from 0. It is not 1.**

I thought that the stream index of Infrared stream always starts from 1 until RealSense L515 is released.
However, that was not the case. It has sensors that start from 1 and sensors that start from 0.

The following pattern makes sense.

RealSense D400 : 0, 1
RealSense L500 : 0

or

RealSense D400 : 1, 2
RealSense L500 : 1

But in reality, it is

RealSense D400 : 1, 2
RealSense L500 : 0

@MartyG-RealSense Do you have special reasons for this specification? Thanks,

---
I have updated [rs_bag2image](https://github.com/UnaNancyOwen/rs_bag2image). it is add supports RealSense L515.

UnaNancyOwen, 2021-05-13:
This is test program for check stream indices of infrared streams.
```cpp
#include <vector>
#include <iostream>
#include <librealsense2/rs.hpp>

int main( int argc, char* argv[] )
{
    rs2::context context;
    rs2::device_list devices = context.query_devices();
    for( rs2::device device : devices ){
        const std::string friendly_name = device.get_info( rs2_camera_info::RS2_CAMERA_INFO_NAME );
        std::cout << "* " << friendly_name << std::endl;
        const std::vector<rs2::sensor> sensors = device.query_sensors();
        for( rs2::sensor sensor : sensors ){
            const std::vector<rs2::stream_profile> stream_profiles = sensor.get_stream_profiles();
            for( rs2::stream_profile stream_profile : stream_profiles ){
                if( stream_profile.stream_type() != rs2_stream::RS2_STREAM_INFRARED ) continue;
                const std::string stream_name = stream_profile.stream_name();
                const int32_t stream_index = stream_profile.stream_index();
                std::cout << "\t - " << stream_name << " (stream_index : " << stream_index << ")" << std::endl;
            }
        }
    }

    return 0;
}
```
```
* Intel RealSense D435
         - Infrared 1 (stream_index : 1)
        - ...
         - Infrared 2 (stream_index : 2)
         - ...
* Intel RealSense L515
         - Infrared (stream_index : 0)
         - ...
```

RealSenseSupport, 2021-05-17:
@ev-mp Could you please clarify the reason to @UnaNancyOwen ? Thanks!