realsenseai/librealsense #9005: How to get textured point cloud?

JimXu1989, 2021-05-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<45>.<?> }                          | 
| Language                            |  {python}                          | 

### Issue Description
How to get textured point cloud? Texture form color image.

MartyG-RealSense, 2021-05-12:
Hi @JimXu1989   As you quote "We'll use the colorizer to generate texture for our PLY", it sounds as though you are referring to the SDK Python example program **export_ply_example**.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

Historically, there have been difficulties in past cases with exporting color to a ply using Python.

The case below has Python scripting shared by a RealSense user whose code exported color to a ply, though without vertex normals.  If lack of normals is not a problem for you then the script may be helpful if it works for you.  It uses the **export_to_ply** export method instead of save_to_ply.

https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

JimXu1989, 2021-05-17:
> Hi @JimXu1989 As you quote "We'll use the colorizer to generate texture for our PLY", it sounds as though you are referring to the SDK Python example program **export_ply_example**.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py
> 
> Historically, there have been difficulties in past cases with exporting color to a ply using Python.
> 
> The case below has Python scripting shared by a RealSense user whose code exported color to a ply, though without vertex normals. If lack of normals is not a problem for you then the script may be helpful if it works for you. It uses the **export_to_ply** export method instead of save_to_ply.
> 
> [#6194 (comment)](https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293)

Thanks a lot for your reply

closed 2021-05-29 by MartyG-RealSense