realsenseai/librealsense #9006: misalignment pointcloud and color frame c#

DirkKramer, 2021-05-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|-------------------------------- |------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version | Win 10  | 
| Platform                                 | PC |
| SDK Version                           |  2.0   | 
| Language                               |  C#   | 
| Segment			|  Robot        | 

Hello, I am using the Realsense d455 camera in combination with c#. My application detects an object and shows the convexhull or boundingbox around the object.  To filter the pointcloud I transform this convexhull to the pointcloud and filter the object out of the pointcloud.  I already succesfully managed to do this using the Azure Kinect, and now trying with realsense products.

### Issue Description

My depthframe/pointcloud and colorframe are slightly misaligned (see code and image).
[image]

```
await Task.Run(() =>
{
    PointCloud pc = new();
    HoleFillingFilter holeFillingFilter = new();
    Align align = new(Stream.Color);
    while (!tokenSource.Token.IsCancellationRequested)
    {
        long begintime = CameraTimer.ElapsedMilliseconds;
        using (FrameSet frames = pipeline.WaitForFrames())
        {
            // Alligned Frameset
            var alignedframes = align.Process<FrameSet>(frames).DisposeWith(frames);

            // Depth Frame
            var depthFrame = alignedframes.DepthFrame.DisposeWith(frames);
            var colorizedDepth = colorizer.Process<VideoFrame>(depthFrame).DisposeWith(frames);
            depthFrame = holeFillingFilter.Process<DepthFrame>(depthFrame).DisposeWith(frames);

            // RGB Frame
            var colorFrame = alignedframes.ColorFrame.DisposeWith(frames);

            // Get Pointcloud vertices.
            var pts = pc.Process<Points>(depthFrame).DisposeWith(frames);
            pts.CopyVertices(vertices);

            // Scale points and add them to sharpdx collection.
            for (int i = 0; i < vertices.Length; i++)
            {
                var vert = vertices[i];
                positions[i] = new Vector3(vertices[i].X, vertices[i].Y, vertices[i].Z) / realsenseDepthscale;
            }
        }
    }
});
```

Also if someone can help to color the points in the pointcloud , that would be greatly appreciated. 
To color the pointcloud in helixtoolkit I need  RGB color foreach point.

MartyG-RealSense, 2021-05-12:
Hi @DirkKramer  There are very limited references available about using librealsense with C# and helixtoolkit to generate a point cloud.  The main reference available is in the link below.

https://github.com/IntelRealSense/librealsense/issues/7362

Given that the mis-alignment between depth and color in your application is slight, it may be worth using the SDK's **Dynamic Calibration** tool to perform a robust calibration of the depth and RGB sensors to check whether a calibration corrects the small mis-alignment.

The Dynamic Calibration tool can be downloaded for Windows from this link:

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

DirkKramer, 2021-05-14:
I already read that issue, but it is not really helpfull.  I just like to know each RGB value of a corresponding point in the pointcloud to show the pointcloud just like the Realsense viewer. I dont think this has anything to to with helixtoolkit specific.

For my misalignment issue, I tried the calibration software, results are still the same.

MartyG-RealSense, 2021-05-14:
Looking over your original script at the top of this case, I note that you seem to be performing align first and then post-processing afterwards.  Is that correct please?

[image]

If it is correct, it is recommended that alignment is performed **after** post-processing in order to help avoid a distortion phenomenon called **aliasing** (jagged lines).  Does re-arranging your script to align after applying the hole-filling filter make a difference to the accuracy of the alignment, please?

DirkKramer, 2021-05-17:
Hey @MartyG-RealSense 

I tried the following script, results are the same. 

```
                        using (frames = Pipeline.WaitForFrames())
                        {
                            // Post Processing
                            frames = holeFillingFilter.Process<FrameSet>(frames).DisposeWith(frames);

                            // Aligner
                            frames = aligner.Process<FrameSet>(frames).DisposeWith(frames);

                            // Depth Frame                             
                            depthFrame = frames.DepthFrame.DisposeWith(frames);

                            // RGB Frame
                            colorFrame = frames.ColorFrame.DisposeWith(frames);

                            // Get Pointcloud vertices.
                            pts = pc.Process<Points>(depthFrame).DisposeWith(frames);
                            pts.CopyVertices(vertices);
                            pts.CopyTextureCoords(textureCoordinates);

                            // Get Colordata 
                            colorData = colorData ?? new byte[colorFrame.Stride * colorFrame.Height];
                            colorFrame.CopyTo(colorData);
                        }
```

For now I solved this issue by applying a transform to all the vertices of -15 in the Y-Direction because the offset seams to be constant. 

But I do want to know where the value of "15" is comming from.

ps. (I managed to get a coloured pointcloud now).

MartyG-RealSense, 2021-05-17:
Whilst I am not aware of an issue that would specifically cause an offset of 15, I am reminded of how there have been past difficulties with depth to color alignment in the C# language.  The links below provide a good set of references about the subject.

https://github.com/IntelRealSense/librealsense/issues/2734
https://github.com/IntelRealSense/librealsense/issues/4719
https://github.com/IntelRealSense/librealsense/issues/5582

MartyG-RealSense, 2021-05-23:
Hi @DirkKramer  Do you require further assistance with this case, please?  Thanks!

DirkKramer, 2021-05-26:
Hi @MartyG-RealSense  I still did not solve this issue,  also i was wrong about it being a constant value of 15. The offset is changing when i change height of camera (Z).  I will look into this in the comming weeks when I have time.

MartyG-RealSense, 2021-05-26:
Thanks very much @DirkKramer for the update.  If it will be a timescale of weeks before you can investigate the issue further, you are welcome to close this case with the **Close Issue** button beneath the comment box and re-open it later if you need to, or start a new case.  Thanks!

closed 2021-05-26 by DirkKramer