realsenseai/librealsense #9012: Running Viewer for the first time, Aborted when switching on camera

innovation-plantation, 2021-05-12:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```
This just happened twice. I plugged in two new RealSense cameras, downloaded and ran viewer, and it offered to and updated firmware on one of them. Then I didn't see any video but noticed it was asking me to turn on the camera (did I add the camera to the list first? I don't remember doing that the first time, but the second time I did) Anyway it seems when I turn on the camera I see one frame and then this message about unhandled exception escaping from a worker thread always seems to be the case now. fatal.

MartyG-RealSense, 2021-05-12:
Hi @innovation-plantation  Which RealSense camera model are your cameras please?

MartyG-RealSense, 2021-05-18:
Hi @innovation-plantation   Do you require further assistance with this case, please?  Thanks!

innovation-plantation, 2021-05-21:
It's OK now - the problem was that one camera was plugged into a USB2 port with a USB3 cable and the other was plugged into a USB3 port with a USB2 cable, so after getting the USB3 cable with the USB port the cameras all worked fine. The USB3 port is not well marked or even color coded on this laptop so I had to look up online which port was what. I suggest that the software be updated to detect when USB2 is being used and notify the user that the port seems to only be talking USB2.

closed 2021-05-21 by innovation-plantation

MartyG-RealSense, 2021-05-21:
Thanks very much @innovation-plantation for the update!  

The RealSense Viewer and Depth Quality Tool programs already highlight the USB mode that the camera is currently being detected as, with a number beside the camera name at the top of the options side-panel.  This is usually **2.1** for a USB 2 connection and **3.2** for a USB 3 connection.

[image]

[image]