realsenseai/librealsense #9013: Input/output error

dheera, 2021-05-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00| 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.4.0-59-generic                         | 
| Platform                                 | x86_64  |
| SDK Version                            |  latest from release | 
| Language                            |  python | 

### Issue Description

This happens randomly and typically after 30-100 minutes of running pyglet_pointcloud_viewer.py.
The pyglet viewer will freeze, and restarting it again causes this issue. The only way I can see to get out of it it is to unplug and replug the USB cable.
```
$ ./pyglet_pointcloud_viewer.py                                                                                                                                
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/usr/bin/xonsh", line 4, in <module>
    main()
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 24019, in main
    _failback_to_other_shells(args, err)
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 23983, in _failback_to_other_shells
    raise err
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 24017, in main
    return main_xonsh(args)
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 24060, in main_xonsh
    run_script_with_cache(
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 3043, in run_script_with_cache
    run_compiled_code(ccode, glb, loc, mode)
  File "/usr/lib/python3/dist-packages/xonsh/__amalgam__.py", line 2948, in run_compiled_code
    func(code, glb, loc)
  File "./pyglet_pointcloud_viewer.py", line 103, in <module>
    pipeline.start(config)
RuntimeError: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error

```

MartyG-RealSense, 2021-05-13:
Hi @dheera   As a starting point in investigating your case, could you monitor memory usage in Ubuntu whilst **pyglet_pointcloud_viewer.py** is running to see whether the amount of available memory is reducing over time (known as a memory leak, which could lead to the program freezing).  

The speed at which memory is consumed (and therefore the time before failure of the program occurs) may vary depending on factors such as the camera resolution and FPS speed being used.  Lower resolution / FPS may mean a longer period of running time before program failure occurs.

Ubuntu's **htop** interface could be used to perform the memory utilization monitoring.

https://vitux.com/how-to-use-htop-to-monitor-system-processes-in-ubuntu-20-04/

MartyG-RealSense, 2021-05-19:
Hi @dheera  Do you require further assistance with this case, please?  Thanks!

dheera, 2021-05-23:
I gave up on it and switched to C++. The Python still seems buggy.

MartyG-RealSense, 2021-05-24:
Thanks very much for the update.  Do you require any assistance with the C++ version of your project, please?

closed 2021-05-30 by MartyG-RealSense