realsenseai/librealsense #9014: Does rs-server support L515?

tsengyushiang, 2021-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |  Win10, ubuntu16.04  | 
| Kernel Version (Linux Only)    |                                | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0                         | 
| Language                            |  official viewer/rs-server                          | 
| Segment			|                  | 

### Issue Description

I want to stream realsense L515 through network by rs-server, but I get error message on both server and client, is there suggestion to fix the issue?

### client error message
[image]

### server error message
[image]

RealSenseSupport, 2021-05-14:
Hi @tsengyushiang 

Unfortunately rs-server hasn't been updated to support L515 so currently no support in this tool for L515.  We would need to investigate the changes it would take to support L515 in rs-server.

closed 2021-05-16 by tsengyushiang

AndreV84, 2021-05-23:
@RealSenseSupport 
What is the status on L515 rs-server?
How to stream over the network with it at all?
It doesn't appear as v4l2 device in /dev/video*
so how to use gstreamer with intels driver in order to stream L515 with gstreamer?

Ali-AliAli-Ali, 2025-01-28:
Could you please provide information about L515 support by rs-server? The question of how to stream over the network with L515 is highly relevant.

| Required Info| |
|--------------|----------------|
| Camera Model | L515 |
| Operating System & Version | Ubuntu 22.04 |
| Kernel Version (Linux Only) | 6.8.0-52-generic |
| Platform | PC |
| SDK Version | 2.47.0 |
| Language | C++ |	

Thanks in advance!

MartyG-RealSense, 2025-01-28:
Hi @Ali-AliAli-Ali  One of my Intel RealSense team members advises at https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-883900014 that the L515 camera model is not supported by rs-server, unfortunately.  As both the L515 camera and the rs-server tool are now retired, L515 support will not be added in future.

It is also worth highlighting that librealsense version 2.47 does not support kernel 6.6 or 6.8.  6.6 is still not supported and 6.8 support was not added until SDK 2.56.2 Beta, which does not support the rs-server tool.

************

There is an alternative method of networking RealSense cameras with small single-board computers that is called **EtherSense**, though it uses Python instead of C++ and I am not aware of a past case of it being tested with an L515.

https://github.com/krejov100/EtherSense

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper