realsenseai/librealsense #9016: jetson

surefyyq, 2021-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435&D435i } | 
| Firmware Version                   | (D435=05.12.08.200; D435i=NULL) | 
| Operating System & Version |   {Linux (Ubuntu 18)}  | 
| Kernel Version (Linux Only)    |  (4.9.140-tegra)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I build librealsense from source by `cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true` and want to run D435 and D435i on jetson TX2 at the same time.

For D435 , command `lsusb` and `rs-enumerate-device` output rightly and the camera works all right;

But when I plug D435i on jetson tx2, `lsusb` command does show the Device Intel Corp. while command `rs-enumerate-device` outputs no device connect.

Why is this and is there any special settings for D435i

Thanks!

MartyG-RealSense, 2021-05-13:
Hi @surefyyq   The key information seems to be that you want to use the D435 and D435i at the same time on Jetson.  The model being D435i instead of D435 is not likely to be the cause of it not working.  There have been past cases in which detection failures have occurred when more than one camera is attached to a Jetson board (though they are capable of multi-camera setups). 

Could you test please whether rs-enumerate-devices can find the D435i if it is the only camera attached and the D435 is unplugged, please?

surefyyq, 2021-05-13:
Hi @MartyG-RealSense Thanks for your quick reply!

When D435i alone is plugged on TX2, `rs-enumerate-devices` outputs `No device detected. Is it plugged in?`

And the realsense-viewer in jetson Ubuntu can't detect D435i either,

But in win10, realsense-viewer works normally for the same D435i.

MartyG-RealSense, 2021-05-13:
Which librealsense SDK version are you using on Jetson please?

surefyyq, 2021-05-14:
Sorry for the late, but how can I check the version of librealsense SDK on jetson?

Before build, I clone it by `git clone https://github.com/IntelRealSense/librealsense.git`.

MartyG-RealSense, 2021-05-14:
If you are installing librealsense from source with CMake then you could choose the version that is built simply by downloading a zip-file of the source code folder from the **Assets** file list of a specific version on the SDK **Releases** page.

https://github.com/IntelRealSense/librealsense/releases

Once the source code zip-file is extracted into a folder then you would go to the root directory and perform **mkdir build && cd build** as usual to prepare that source-code folder for building the particular librealsense version in that folder.

I would not recommmend installing a librealsense version as old as 2.18.0 though.  And I am not aware of a reason why using TX2 with D435i on a modern version such as 2.39.0 would prevent the camera from being detected.

If you prefer to build from source with CMake then you may increase you chances of success if you use instructions in the link below that have been consistently successful with Jetson boards.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

I have also found though that building with **Debian packages** on TX2 instead of source code has also been successful.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

surefyyq, 2021-05-16:
Problem solved, thank you!

According to https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977 and https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049, the `No device detected. Is it plugged in?` issue may be related to several reasons including cmake version, librealsense version and cmake build parameters. 

I tested several experiments with controlled variable and it turns out that cmake and librealsense version are not the key.

With all other conditions remain the same, my original way of  building from source: `cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true`  works for D435 but doesn't work for D435i, 
but the way `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true` works for both D435 and D435i !

Don't know why, but I'm closing this issue~

closed 2021-05-16 by surefyyq

MartyG-RealSense, 2021-05-16:
Great news @surefyyq that you found a method that works for you - thanks very much for the update!