realsenseai/librealsense #9018: How to set autoexposure limit in python code

bartonp2, 2021-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  5.4.0-73-generic     | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0  | 
| Language                            | python   | 
| Segment			|  Robot     | 

### Issue Description
While I can set the autoexposure limit in the realsense-viewer I have been unable to do this in python. I have tried setting it in the json config file with the '`controls-autoexposure-limit`' keyword but get the following error:
`RuntimeError: controls-autoexposure-limit key is not supported by the connected device!`

I have also tried setting it with the set_option method:
```
sensor = pipeline.get_active_profile().get_device().query_sensors()[0]
sensor.set_option(rs.option.auto_exposure_limit, 100)
```
But the error is similar:
`RuntimeError: object doesn't support option #85`

But since the option works in the realsense-viewer I must be doing something wrong. Could you maybe point out my mistake or give some example code?

MartyG-RealSense, 2021-05-13:
Hi @bartonp2  Another RealSense Python user recently experienced the same **RuntimeError: object doesn't support option #85** error when attempting to convert the **RS2_OPTION_AUTO_EXPOSURE_LIMIT** instruction to Python by using **sensor.set_option(rs.option.auto_exposure_limit, 100)**.

https://github.com/IntelRealSense/realsense-ros/issues/1821#issuecomment-827342776

RS2_OPTION_AUTO_EXPOSURE_LIMIT is listed in the official C++ documentation:

https://intelrealsense.github.io/librealsense/doxygen/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2ac11001971364b90b542941e51ebb911a

I could not find an equivalent instruction in the official pyrealsense2 documentation though.

https://intelrealsense.github.io/librealsense/python_docs/index.html

The RealSense Viewer is a C++ application, so it makes sense that auto-exposure limit would be accessible.

I attempted to set auto-exposure limit in the Viewer and then export a json file from the Viewer but the auto-exposure limit value was not exported into the json.

I will seek advice from Intel to confirm whether auto-exposure limit is currently exclusive to C++.

MartyG-RealSense, 2021-05-13:
After consultation with Intel, an official bug report has been filed with Intel regarding the issues in the above comment.  This case should be kept open whilst the bug report is active.  Thanks again!

bartonp2, 2021-06-21:
Has there been any progress on this issue?

MartyG-RealSense, 2021-06-21:
@bartonp2  I have no further information to provide at this time.

MartyG-RealSense, 2021-08-23:
Hi @bartonp2   Intel have informed me that that the **auto_exposure_limit** instruction is already supported in pyrealsense2.  There does not seem to be an entry for it in the pyrealsense2 documentation but a Python script recently made use of it, so I have posted a link to that script reference below.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001050241/comments/1500000948182

bartonp2, 2021-08-23:
Hi @MartyG-RealSense, thanks for the suggestion. But unfortunatly this does not work for me, as noted in the issue description, and results in the runtime error:
`RuntimeError: object doesn't support option #85`

MartyG-RealSense, 2021-08-23:
Intel did also suggest that if you installed with a Python **wheel** then the problem that you are having could be related to that.

MartyG-RealSense, 2021-08-29:
Hi @bartonp2  Do you require further assistance with this case, please?  Thanks!

bartonp2, 2021-08-30:
Hi MartyG,

I am not working on this project at the time being and do not require further assistance for the time being.
Thank you!

MartyG-RealSense, 2021-08-30:
Thanks very much @bartonp2 for the update!  This case should be kept open for the moment as there is an active Intel bug report associated with it, but you do not need to take any further action.  Thanks again   :)

MartyG-RealSense, 2021-09-06:
Hi again @bartonp2   The Intel bug report that was associated with this case is now closed, so this case can be closed too.  Please feel free to re-open this case or create a new one if you return to your project at a future date and require further assistance with this particular question.  Good luck!

closed 2021-09-06 by MartyG-RealSense