realsenseai/librealsense #9023: configure CMake

rain0902, 2021-05-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) } | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
An error occurred when configuring CMake.As follows:

CMake Error: Cannot open file for write: C:/Program Files/Realsense/librealsense-master/CMakeCache.txt.tmpa9ecf
CMake Error at C:/Program Files/CMake/share/cmake-3.20/Modules/CMakeDetermineSystem.cmake:203 (file):
  file failed to open for writing (No such file or directory):

    C:/Program Files/Realsense/librealsense-master/CMakeFiles/CMakeOutput.log
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error: Could not open file for write in copy operation C:/Program Files/Realsense/librealsense-master/CMakeFiles/3.20.2/CMakeSystem.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at C:/Program Files/CMake/share/cmake-3.20/Modules/CMakeDetermineSystem.cmake:215 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.18363.
CMake Error at CMakeLists.txt:3 (project):
  Failed to make directory:

    C:/Program Files/Realsense/librealsense-master/CMakeFiles/3.20.2

Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: C:/Program Files/Realsense/librealsense-master/CMakeCache.txt.tmpec537
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. C:/Program Files/Realsense/librealsense-master/CMakeCache.txt
CMake Error: : System Error: Permission denied
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. C:/Program Files/Realsense/librealsense-master/CMakeCache.txt
CMake Error: : System Error: Permission denied

Why does this problem occur? How to solve it？Thank you!
And I don’t know how to do the first step of Building from Source. Is there a detailed explanation?

MartyG-RealSense, 2021-05-14:
Hi @rain0902   As you are using Windows, you may not need to build librealsense with CMake from source as there is an automated installer file for Windows.  It can be found in the **Assets** file list at the bottom of the information listing for each SDK version on the **Releases** page.

https://github.com/IntelRealSense/librealsense/releases

[image]

If you do need to build librealsense on Windows from source code using CMake though then the guide in the link below can be followed.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

You should be able to skip the section of the guide about enabling metadata, as you can do so in the **RealSense Viewer** tool instead after the SDK has been installed.

In regard to the CMake error that you wrote about if you still require an answer to that question after reading the links above then could you provide information please about the steps that you took before the error occurred?  Thanks!

rain0902, 2021-05-17:
Hello, first of all thank you for your reply. I‘m sorry, I didn’t make it clear in the previous letter.

I want to controll the RealSense Cameras with Matlab R2020b in Windows 10. This is the link：[](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#getting-started-with-realsense-sdk20-for-matlab)

But I don't know how to do the first step of building from Source, which is "Download the Git repository. Opt in the Matlab wrapper for LibRs with cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON when configuring CMake.".  Is there a detailed explanation?Please.Thank you very much!

rain0902, 2021-05-17:
Hello, first of all thank you for your reply. I‘m sorry, I didn’t make it clear in the previous letter.

I want to controll the RealSense Cameras with Matlab R2020b in Windows 10. This is the link:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#getting-started-with-realsense-sdk20-for-matlab

But I don't know how to do the first step of building from Source, which is "Download the Git repository. Opt in the Matlab wrapper for LibRs with cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON when configuring CMake.".  Is there a detailed explanation?Please.Thank you very much!

MartyG-RealSense, 2021-05-17:
Hi @rain0902   You may find it easier to install the MATLAB wrapper as part of the automated RealSense SDK installer file's installation process.

1.  Go to the SDK **Releases** page and scroll down the version listings to the bottom of the SDK version that you want to install.

https://github.com/IntelRealSense/librealsense/releases

2.  At the bottom of each version listing is an **Assets** link with a triangle beside it. Left-click on the triangle to expand open a file list.  Left-click on the file link that begins with the name **Intel.RealSense.SDK-WIN10** to begin downloading the large installer file for the SDK at the bottom of your browser window.

[image]

[image]

If you want to keep using a particular version and do not want to upgrade to a newer one then simply select the version that you were already using, as the installer will overwrite the previous installation.

3.  When the download is complete, left-click on the download at the bottom of the window to launch the downloaded installer file and run the installation process.

[image]

4.  Left-click the **Next** button on the installer window and on its next page, check that **Matlab developer package** has a tick in the box beside it to ensure that the MATLAB wrapper will be installed as part of the overall SDK installation process.

[image]

If there is a tick beside it then click **Next** to continue the installation process and keep simply clicking **Next** until it is shortly completed.

5.  Once the SDK is installed on your Windows computer, navigate to the directory location below to find a MATLAB wrapper folder called **+realsense**

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > matlab**

[image]

6.  Add the **+realsense** folder to the MATLAB search path using the guidance in the link below.

https://github.com/IntelRealSense/librealsense/issues/2400

Alternatively, you can copy and paste this folder to a location on your computer where MATLAB can find it.

rain0902, 2021-05-18:
Hello, thank you very much. I have completed these six steps. 

But I still don’t know how to do the intermediate steps before the guidance in https://github.com/IntelRealSense/librealsense/issues/2400
In the other words, I haven't finished building the wrapper from sources, and I don't have a directory called +realsense with mex and matlab wrapper. 

I found the link https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#getting-started-with-realsense-sdk20-for-matlab
But I don’t know how to do the first step of building from Source, which is "Download the Git repository. Opt in the Matlab wrapper for LibRs with cmake .. -D... -DBUILD_MATLAB_BINDINGS:BOOL=ON when configuring CMake.". Is there a detailed explanation?Please.Thank you very much!

closed 2021-05-18 by rain0902

MartyG-RealSense, 2021-05-18:
You do not need to build the MATLAB wrapper from source code if you have installed it with the automated SDK installer.  The Getting Started instructions offer the choice of installing with source code **or** the Windows installer.  You do not need to perform both.

Are you able to find the SDK's folder location on your computer by right-clicking on the RealSense Viewer launch shortcut icon on your Windows desktop and selecting the **Open file location** option from the menu, please?

rain0902, 2021-05-18:
Yes, I can find it, and it is in C:\Program Files (x86)\Intel RealSense SDK 2.0\tools