realsenseai/librealsense #9024: Intel Realsense SR300 with Jetson Nano produces incorrect depth values in Ubuntu

bula0422, 2021-05-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (3.21.0.0) | 
| Operating System & Version |    Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (4.9.14)                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  {2.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Computer Vision }                          | 

### Issue Description
Hi, I have the same issue with https://github.com/IntelRealSense/librealsense/issues/5845.
I have tried SDK in both 2.31 and 2.41 that can't get correct depth value.
Depth value stays about 0.1 meters.
The issue does not exist in Win OS.
Does anyone can get correct depth value in Jetsno Nano? Which version of jetpack and SDK?

MartyG-RealSense, 2021-05-14:
Hi @bula0422  The subject of depth problems with SR300 when using **CUDA** (as Nvidia Jetson boards do) is a known issue.  A RealSense user kindly contributed a Pull Request (PR) a few weeks ago to address the issue.

https://github.com/IntelRealSense/librealsense/issues/8897
https://github.com/IntelRealSense/librealsense/pull/8896

The pull page states that it was merged into the **development** branch of the librealsense SDK 9 days ago at the time of writing this.  As SDK 2.45.0 was released 12 days ago at the time of writing this, the fix may therefore be included in the next SDK version after the current one.

Alternatively, the development branch of the SDK can be accessed that the fix has been merged into can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/development

MartyG-RealSense, 2021-05-21:
Hi @bula0422  Do you require further assistance with this case, please?  Thanks!

bula0422, 2021-05-24:
Hi @MartyG-RealSense  Thanks for your reply, so I just need to change the address as https://github.com/IntelRealSense/librealsense/tree/development for git clone command in the https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh right?

MartyG-RealSense, 2021-05-24:
Are you referring to the git clone instruction in the link below, please?

https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh#L54

If you are, and you are aiming to target the librealsense **development** branch instead of the **master** branch, my research indicates that a specific **branch** can be specified in the git clone instruction with **-b <branch name>**, followed by the remote git URL.  For example:

**git clone -b development https://github.com/IntelRealSense/librealsense.git**

There is also the option of course of waiting for the release of the next librealsense version after 2.45.0.

MartyG-RealSense, 2021-05-25:
There is not a fixed schedule for new librealsense SDK releases but typically they arrive about every 30 days.  At the time of writing this, the current 2.45.0 SDK has been released for 23 days.

MartyG-RealSense, 2021-06-06:
Hi @bula0422   This case may be related to a past issue of depth value inaccuracies with SR300 on Jetson due to CUDA.  A fix for the problem was created and may be included in the next librealsense version after the current 2.45.0 SDK.

https://github.com/IntelRealSense/librealsense/issues/6373#issuecomment-841160923

MartyG-RealSense, 2021-06-07:
RealSense SDK 2.47.0 is now available and includes the previously mentioned SR300 fix.

[image]

MartyG-RealSense, 2021-06-13:
Hi @bula0422   As a fix for SR300 incorrect depth was released in SDK **2.47.0**, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2021-06-18:
Case closed due to no further comments received and a fix for incorrect depth with CUDA released in SDK 2.47.0.

closed 2021-06-18 by MartyG-RealSense