realsenseai/librealsense #9027: Algorithm for mesh triangulation from point cloud

Dhelio, 2021-05-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Win10 / Android 10  |
| Platform                                 | PC / Custom board  |
| SDK Version                            | 2.44.0 | 
| Language                            | Unity | 
| Segment			|  AR | 

Hello,

I've spent a couple of days documenting about a mesh generation algorithm starting from unorganized points in a point cloud.

My objective is to reconstruct a mesh from a point cloud obtained from a realsense D435i, preferably in realtime. Precision is not essetial as of right now, but speed is. The final **very very rough** result I'd like to achieve is something similar to Microsoft's spatial mapping ( [seen here](https://www.youtube.com/watch?v=zff2aQ1RaVo))

I've read about Delaunay's triangulation, but it works well mostly in 2D, while being too slow in 3D (paper here: https://www.kiv.zcu.cz/site/documents/verejne/vyzkum/publikace/technicke-zpravy/2002/tr-2002-02.pdf); plus it probably doesn't achieve what I was going for (sources: https://github.com/QThund/ConstrainedDelaunayTriangulation and https://github.com/Scrawk/Hull-Delaunay-Voronoi). They generate mostly closed shapes;

I've read about the ball-pivoting algorithm (here: https://github.com/Renge-Games/UnityBPA), but it's not what I was looking for: too slow for a realtime application.

I've read the paper that Microsoft probably used to solve this kind of problem (here: http://hhoppe.com/thesis.pdf), but its application isn't immediate as I'd hoped.

So I changed approach. What if there was an external service to do this in?
From the issue https://github.com/IntelRealSense/librealsense/issues/6225 I've read that there are application that do this, such as CloudCompare (here: https://www.danielgm.net/cc/) or PointCLoud (here: https://pointclouds.org). But  they probably don't work on Android, which is my target platform.

What I wanted to know is this: is there any algorithm - already implemented or not - or external service that alows to have a mesh - any mesh, even a dense one - in realtime that works on Android ?

MartyG-RealSense, 2021-05-14:
Hi @Dhelio   I researched your question carefully.  Looking at the example of Microsoft's spatial mapping video, it looks as though you wish to map the contents of a room so that virtual content can interact with it (such as projecting onto surfaces or colliding a virtual object with the surface).  Is that correct, please?

Intel demonstrated something similar in 2016 with a prototype 'merged reality' headset called **Project Alloy** that could scan the scene and take account of real-world objects so that virtual and real content could mix together (hence 'merged reality').  The YouTube video in the link below demonstrates Project Alloy in action on a convention event stage.

https://www.youtube.com/watch?v=Be-MJxeCpvU&t=101s

The Project Alloy headset contained a RealSense ZR300 stereo camera, the predecessor of the RealSense D435 stereo camera model.

In regard to how you might implement such a system in Unity: a RealSense user developed a RealSense point cloud system for the Android-based **Oculus Quest** headset that combined the Quest with a D435i camera and published a project describing how to create it.

https://github.com/GeorgeAdamon/quest-realsense

https://github.com/IntelRealSense/librealsense/issues/4155

As camera data in the RealSense Unity wrapper is visually represented by being mapped onto objects via a material, I wonder whether this may suit your project requirements.

Dhelio, 2021-05-14:
>  it looks as though you wish to map the contents of a room

Not exactly. I want to map the room itself.

MartyG-RealSense, 2021-05-14:
The commercial software **DotProduct Dot3D Scan** is compatible with 400 Series cameras and can be used on Android to produce point-cloud scans.  

https://www.dotproduct3d.com/dot3dscan.html

There is also a **Dot3D Pro** version for scanning larger scenes.  This pro version has support for a DotProduct product called **InfiPoints DP** that can be used on Android for translating data to CAD.

https://www.dotproduct3d.com/infipointsdp.html

Overall though, the requirement for Android means that it may be difficult to fulfill the project goals.  Would it be possible to create the project on a Windows tablet instead?

sam598, 2021-05-14:
@Dhelio I would strongly recommend using a DepthMap to TSDF to Mesh pipeline instead of pointclouds, similar to the famous KinectFusion technique. This is most likely what every current real-time room mapping technique is doing under the hood.

Meshing voxels are much easier to process and parallelize than pointclouds. Using a TSDF you can easily merge multiple depth maps to increase coverage, accuracy, and resolution.

The Open3D project maintained by Intel has examples of how to do this, including RealSense examples.
https://github.com/intel-isl/Open3D

Dhelio, 2021-05-14:
Thank you @sam598 , this is probably what I was looking for; I'll look into it on monday.

closed 2021-05-14 by Dhelio