realsenseai/librealsense #9028: Unable to access depth stream on L515

N-Chase, 2021-05-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515
| Firmware Version                   | 01.05.05.00
| Operating System & Version | Win10
| Platform                                 | PC
| Language                               |  python 

### Issue Description
Hello,
the depth/infrared system of our L515 suddenly stopped working. It's no longer possible to acquire any depth information from the device via RS viewer or the python infrastructure. I have tested it on a Windows and a Linux machine, but to no avail. The RGB component seems to be functional though and I can successfully connect to the stream.

In RS viewer i set the log-level to DEBUG and this seems to be the critical part the keeps on failing:

----------------------------------------------------------------------------------------------------

14/05 11:01:45,277 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,277 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,277 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,278 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,278 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,278 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 14/05 11:01:45,280 ERROR [11916] (types.h:315) _reader->ReadSample(...) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,280 DEBUG [1392] (win-helpers.cpp:87) owner->_reader->ReadSample(dwStreamIndex, 0, nullptr, nullptr, nullptr, nullptr) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,281 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,282 DEBUG [11916] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0x80070018: "Das Programm hat einen Befehl ausgegeben, aber die Befehlslänge ist falsch."
 14/05 11:01:45,289 DEBUG [11916] (types.cpp:78) recoverable_exception: 
Failed to resolve the request: 
	Format: Z16, width: 1024, height: 768
	Format: Y8, width: 1024, height: 768
	Format: RAW8, width: 1024, height: 768

Into:
	Formats: 
	 Z16
	 Y8
	 RAW8

 14/05 11:01:45,289 ERROR [11916] (l500-depth.cpp:714) Exception caught in l500_depth_sensor::open
----------------------------------------------------------------------------------------------------

Does anyone have a clue what could be going on here? Is this some sort of HW defect?

Thanks in advance,
Norman

RealSenseSupport, 2021-05-14:
Hi @N-Chase 

Can you outline how you were using the camera and in what environment?

You can also submit a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

N-Chase, 2021-05-17:
The camera was used to calculate the position of a robotic system. It was positioned on a table pointing at the system at a distance of roughly 30 Centimeters. The environment was not too spectacular either: Office-type location with a low relative humidity (20 %) and a mean temperature of 24 degree Celsius.

RealSenseSupport, 2021-06-10:
@N-Chase 

Did you submit a ticket at the previously provided link?  Did you RMA the unit?

N-Chase, 2021-06-12:
Yes. I did submit a ticket and have been told to return the device. A colleague shipped the unit back to you roughly 2,5 weeks ago, but as far as I know we're still waiting for an acknowledgment and replacement camera.

RealSenseSupport, 2021-06-18:
Hopefully the replacement does come soon.

closed 2021-06-18 by RealSenseSupport