realsenseai/librealsense #9032: Fehler Intel Realsense

DieterWagner28, 2021-05-15:
| | |
|---|---|
|**librealsense**|2.44.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2021-05-15:
Hi @DieterWagner28  The version of the RealSense Viewer that is bundled with the RealSense SDK installation has a requirement of a 6th Generation Intel Core processor or newer because of its OpenVINO functions.  Your E5-2643 is a 2nd generation processor.

You can though download a standalone version of the RealSense Viewer named **Intel.RealSense.Viewer.exe** from the **Assets** file list on the SDK **Releases** page that does not have this requirement because it does not contain the OpenVINO functionality and so will work with older processors.  The rest of the SDK should also work fine on an old processor, as the processor requirement is only on the bundled Viewer program and the 400 Series cameras can work with any Intel or Arm processor.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

Once the standalone Viewer is downloaded, you should double-leftclick on Intel.RealSense.Viewer.exe in the folder that you downloaded it to to launch the Viewer.  Do not use the launch shortcut on the Windows desktop, as that links to the bundled Viewer version with the processor requirement.  

If you need a launch shortcut, you can delete the desktop Viewer shortcut and create a new shortcut to the standalone Viewer's .exe file by right-clicking on the file and selecting the **Create shortcut** menu option.

[image]

DieterWagner28, 2021-05-15:
Vielen Dank,

es hat alles prima geklappt!

 

MfG

Dieter Wagner

 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

 <https://user-images.githubusercontent.com/41145062/117535805-54390680-afef-11eb-86be-4f2842535be2.png>

MartyG-RealSense, 2021-05-15:
Excellent news!  Thanks very much for the update. 

Hervorragende Nachrichten!  Vielen Dank für das update.

closed 2021-05-21 by MartyG-RealSense