realsenseai/librealsense #9036: Is there a way to remove points too far away when I save a .plt file?

JimXu1989, 2021-05-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<45>.<0> }                          | 
| Language                            |  python                          | 

### Issue Description
I use the following code to save a .ply file, while, there are so many far away points which I do not need
pc.map_to(color_frame)
points = pc.calculate(aligned_depth_frame)
points.export_to_ply('./ply_files/'+ str(filelen) + ".ply", color_frame)
As the pic shows:
[image]

Is there any way to clip the 3d points?

MartyG-RealSense, 2021-05-17:
Hi @JimXu1989   You could use a **threshold filter** post-processing filter in Python to set a maximum depth distance so that the far-off points are excluded from the point cloud before it is imported to ply.  An example of such a script is in the link below.

https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301

Alternatively, you could import the ply file into the free **MeshLab** software tool and edit the data there.

https://www.meshlab.net/

JimXu1989, 2021-05-20:
> Hi @JimXu1989 You could use a **threshold filter** post-processing filter in Python to set a maximum depth distance so that the far-off points are excluded from the point cloud before it is imported to ply. An example of such a script is in the link below.
> 
> [#8170 (comment)](https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301)
> 
> Alternatively, you could import the ply file into the free **MeshLab** software tool and edit the data there.
> 
> https://www.meshlab.net/

Hi,
Thanks a lot for your reply!
When I fellow your reply,I met an error :
frames = pipeline.wait_for_frames()
# frames.get_depth_frame() is a 640x360 depth image
frames = threshold_filter.process(frames)
# Align the depth frame to color frame
aligned_frames = align.process(frames)

aligned_frames = align.process(frames)
TypeError: process(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.align, frames: pyrealsense2.pyrealsense2.composite_frame) -> pyrealsense2.pyrealsense2.composite_frame

Invoked with: <pyrealsense2.pyrealsense2.align object at 0x7f8583f3c8b0>, <pyrealsense2.frameset Z16 BGR8 #7>

MartyG-RealSense, 2021-05-20:
It is difficult to debug without seeing the whole script.  Would it be possible to post your full script in the comments please?

For example, have you set up **align_to** in your script 

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L51-L52

MartyG-RealSense, 2021-06-03:
Case closed due to no further comments received.

closed 2021-06-03 by MartyG-RealSense

jianjuan, 2021-08-23:
Hi, I met the same question, I want to filter the background on the pointcloud, here are my codes:
`import pyrealsense2 as rs
print("Environment Ready!")

DEBUG = False
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
pc = rs.pointcloud()
points = rs.points()
thresholder = rs.threshold_filter()
thresholder.set_option(rs.option.min_distance, 0.4)
thresholder.set_option(rs.option.max_distance, 0.9)
# Start streaming
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)

# Wait for a coherent pair of frames: depth and color
# Skip the first 300 frames to set the auto exposure time
for x in range(100):
    pipeline.wait_for_frames()

frames = pipeline.wait_for_frames()
frames = thresholder.process(frames)

# Align the depth frame to color frame
aligned_frames = align.process(frames)
# Get aligned frames
aligned_depth_frame = aligned_frames.get_depth_frame()
aligned_color_frame = aligned_frames.get_color_frame()
aligned_color_profile = rs.video_stream_profile(profile.get_stream(rs.stream.color))
color_intrinsics = aligned_color_profile.get_intrinsics()`

And I got this error:
[image]
It seems that threshold_filter has changed some properties of 'frames'? Can you help me with solving this?Thanks! @MartyG-RealSense

MartyG-RealSense, 2021-08-23:
Hi @jianjuan  You are performing alignment **after** post-processing (the threshold filter) which is Intel's recommendation, so that part looks fine.

In your frame skip routine, I believe that the line **pipeline.wait_for_frames()** could be deleted because the **frames = pipeline.wait_for_frames()** line will handle it correctly.

[image]

If you are only setting a maximum depth rendering distance and do not need to define a minimum distance too then the **clipping distance** method demonstrated in the SDK's **align-depth2color** Python alignment sample program may be able to perform this function instead of a threshold filter.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L52

jianjuan, 2021-08-24:
Thanks for your reply! However, my command is to clip distance from minimum distance to maximum distance, you see, in my applications, close objects are obstruct for my interested object. And I see that "realsense_viewer" can do threshold filter to obtain my ideal pointcloud, so is there a chance to do so in my code? Or, after `thresholder.process()`, is there any operation to convert data type of `frame`?

MartyG-RealSense, 2021-08-24:
The threshold filter can be accessed in Python.  https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301 has an example of a script for doing so that you can compare to your own code.