realsenseai/librealsense #9037: l515 confidence maps

chapa17, 2021-05-17:
I am using L515 intel realsense sensor.

Ubuntu 18.04
Python 3.6

I am trying to extract depth, color and confidence values from a recorded *.bag file using realsense-viewer. I need to get the confidence values for each pixel in a frame. I am not sure if the confidence stream provides the same or not. 

```
pipeline = rs.pipeline()
config = rs.config()

rs.config.enable_device_from_file(config, '*.bag')

config.enable_stream(rs.stream.color)
config.enable_stream(rs.stream.depth)
config.enable_stream(rs.stream.confidence)
```
Does stream.confidence provide confidence maps for each pixel? How do I extract confidence values after enabling the confidence stream?

Thank you in advance!!

goloste, 2021-05-17:
I'm currently looking for the same problem, I want to display a depth confidence map along with the depth map.

The only thing I could understand is that for each depth pixel, condifence data is a 4 bit value, to be retrieved from the confidence stream. If i understood correctly confidence stream comes in a ['CNF4'](https://static.lwn.net/kerneldoc/media/uapi/v4l/pixfmt-cnf4.html) format, this would be a 16 bit frame where first 8 bits and last 8 bits are associated respectively to a value 'n', each 8 bit word contains  (in little endian) the confidence value for pixel (n+1)*2 (bits 7 - 4) and pixel n*2 (bits 3 - 0).

[image]

According to https://github.com/IntelRealSense/librealsense/issues/7029#issuecomment-669949534
The confidence format is accessible  as RAW8.

so i tried to configure the confidence stream as
`config.enable_stream(rs.stream.confidence, 1024, 768, rs.format.raw8, 30)`

and open the pipeline
`pipeline.start(config)`

This method seems to effectively open the confidence stream (as pipeline.start() executes with no errors)

Same as you I couldn't understand how to access the confidence stream.

Please let me know if the information above turns out useful and you figure out a method.

RealSenseSupport, 2021-06-10:
Will you be needing further help with this?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2021-07-28 by RealSenseCustomerSupport

arossbach10, 2022-08-22:
The question is still open.
How to access the data in Python after successfully opening the confidence stream?

I would expect something like:
     pyrealsense2.composite_frame  get_confidence_frame(...)
But this doesn't exist.

How to access the data?