realsenseai/librealsense #9039: I'm running the d435 camera using my Jetson nano(ubuntu 18) i tried running the terminal code for adding the servers to the list of repositories and I ended up with this

miku54, 2021-05-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  ubuntu18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Err:27 http://realsens-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release
404 Not Found [IP: 52.216.134.75 80]
Hit:28 https://repos.codelite.org/ubuntu focal InRelease
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
E: The repository 'http://realsens-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
N: Skipping acquire of configured file 'universe/binary-arm64/Packages' as repository 'https://repos.codelite.org/ubuntu focal InRelease' doesn't support architecture 'arm64'

MartyG-RealSense, 2021-05-18:
Hi @miku54  Could you confirm please which installation method you are using with your Jetson please?

Is it the Debian packages referenced in the **Linux Distribution** installation instructions:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Or the package installation procedure for Jetson boards:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

miku54, 2021-05-18:
Hello, I used the first method you mentioned for installation.

MartyG-RealSense, 2021-05-18:
The Debian packages in the first method **distribution_linux** are not compatible with devices with Arm processors like Nvidia Jetson as these packages require an x86 / x64 processor (like the ones in desktop and laptop computers).

You can install from packages on Jetson using the second link.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Alternatively, you can build librealsense on Nano from source code using CMake with the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

miku54, 2021-05-19:
Thank you, because I chose the wrong installation method, which has been solved at present.

MartyG-RealSense, 2021-05-19:
That is great to hear @miku54 - thanks very much for the update!

closed 2021-05-25 by MartyG-RealSense