realsenseai/librealsense #9040: Release to apt repository for ubuntu focal / arm64

furushchev, 2021-05-18:
Hello,

I'm using Ubuntu focal (20.04) on my computer whose architecture is arm64 and found that there is no released packages for the environment in the apt repo:
https://librealsense.intel.com/Debian/apt-repo

(FYI, there is released packages for Ubuntu bionic (18.04) / arm64)

Is there any issue that blocks the release?

Maybe related to:
https://github.com/IntelRealSense/librealsense/issues/8782

MartyG-RealSense, 2021-05-18:
Hi @furushchev   Could you provide details please about the Arm-equipped computer that you are using?  For example, is it an Apple Mac with the Apple Silicon M1 processor chip?  Thanks!

furushchev, 2021-05-18:
I'm using Raspberry pi 4, Jetson Xavier and TX2 for the real hardware.

```

$ apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
$ add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo focal main" -u
Get:1 https://librealsense.intel.com/Debian/apt-repo focal InRelease [2400 B]
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease                         
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Fetched 2400 B in 1s (1631 B/s)
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
```

MartyG-RealSense, 2021-05-18:
If you are using the **Linux Distribution** instructions in the link below then those packages are not compatible with Arm processors as they require x86 / x64 processors like those in desktop and laptop computers.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Packages can be installed on Jetson using the instructions in the link below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

There is also the option of building librealsense from source code with CMake on Jetson using these instructions:

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

If you are using Ubuntu on Raspberry Pi then the guide below may be helpful:

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

furushchev, 2021-05-18:
Thank you for the instruction.

So my question is, there **is** released packages in the `https://librealsense.intel.com/Debian/apt-repo` repo  for bionic / arm64 which is not supported according to the instructions.
Do you have any plan to add arm64 to the list of supported architecture and add packages for the focal / arm64?

Since many of arm64 hardwares are less powerful than amd64 and it is really helpful if we can install librealsense binaries by using just running `apt-get install` instead of building it from source which will take very long time.

MartyG-RealSense, 2021-05-18:
I am not aware of an official arm64 alternative to the Jetson page's instructions for installing from packages or of upcoming plans to change this.

The website **JetsonHacks** offers a method of installing on Jetson from the librealsense Debian respository using a script called **installLibrealsense.sh**

https://github.com/jetsonhacks/installRealSenseSDK#installlibrealsensesh

furushchev, 2021-05-19:
Thanks for sharing the information.
I know how to install the packages using apt-get install, I'd like to ask maintainers of the apt repository of this library (maybe @dorodnic ? )  if they have a plan to support the arm64 architecture.

MartyG-RealSense, 2021-05-20:
It will be helpful if you can provide more information please about the kind of support that you are asking for regarding arm64.  For example, do you wish the X86/AMD64 packages in the Linux distribution instructions to work with Arm devices too?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

furushchev, 2021-05-25:
> do you wish the X86/AMD64 packages in the Linux distribution instructions to work with Arm devices too?

@MartyG-RealSense 
Yes.
Currently, we can install the library following the Linux distribution instructions on environment below:

|                | i386 | amd64 | arm64 |
|----------------|------|-------|-------|
| xenial (16.04) | Yes  | Yes   | Yes   |
| bionic (18.04) | Yes  | Yes   | Yes   |
| focal (20.04)  | Yes  | Yes   | No    |

I'd like you to support focal / arm64 environment as same as other environments.

ref:
https://librealsense.intel.com/Debian/apt-repo/dists/xenial/InRelease
https://librealsense.intel.com/Debian/apt-repo/dists/bionic/InRelease
https://librealsense.intel.com/Debian/apt-repo/dists/focal/InRelease
Only in focal release, packages for arm64 is missing.

MartyG-RealSense, 2021-05-25:
Thanks very much @furushchev for the clarification.  I will seek advice from Intel about your question.  Thanks for your patience!

furushchev, 2021-06-07:
@MartyG-RealSense @dorodnic Kindly ping to maintainers...

MartyG-RealSense, 2021-06-07:
I have followed up on this enquiry recently but I do not have information to provide yet.  Thanks for your patience.

MartyG-RealSense, 2021-06-08:
Hi @furushchev  After discussion with Intel, I have filed an official feature request asking for consideration of adding Arm64 apt repo support for Focal.  This case should be kept open whilst that feature request is active.  You do not need to take any further action though.  Thanks for the report!

furushchev, 2021-07-07:
@MartyG-RealSense Any update on this issue? It has passed about one month since the last update...

MartyG-RealSense, 2021-07-07:
@furushchev   I do not have a time schedule for this feature that I can provide at present, though Intel have been looking at it.

MartyG-RealSense, 2021-11-03:
Hi @rosskidson  I do not have information regarding any future plans for adding Arm support for 20.04 to the Dynamic Calibration tool.

If you are using C++ in your robot calibration process (as suggested by your interest in the Dynamic Calibration tool's 'DSDynamicCalibration.h' file)  then you should still be able to access the On-Chip system of calibration that is embedded in the camera firmware, as the On-Chip system can be interfaced with using C++ code.  That way, you should be able to program controls for read and write of calibration parameters immediately without having to wait for 20.04 Arm support in the Dynamic Calibration tool.

andresgalaviz, 2022-07-02:
Any updates on arm64 support for 20.04?

MartyG-RealSense, 2022-07-02:
Hi @andresgalaviz  I will make enquiries and let you know when I have information about Arm64 support for 20.04.  Thanks very much for your patience!

JHPHELAN, 2022-07-02:
@MartyG-RealSense @andresgalaviz 
I second that!!  The ARM community is a huge market that should be courted by Intel.
Also especially the ROS and moreso the ROS2 community to provide nodes for the whole RealSense family.
So far I've found librealsense frustrating to work with and have had to look at alternative sensors.
A plug-and-play "apt get" download would be MOST appreciated!!!!!
JHP

MartyG-RealSense, 2022-07-06:
Hi everyone, there is a draft kernel patch at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 that can be tried out.  It adds initial support for kernel 5.13 / 5.15 and provides a pre-built package for testing.

[image]

Chrisx19, 2022-08-10:
Hello I am also having the same problem installing Realsense SDK 2.0 on Ubuntu 20.04 arm64, aarch64. Would it be possible to have a great source I follow up?

MartyG-RealSense, 2022-08-10:
Hi @Chrisx19   A RealSense user with Pi 4 and Ubuntu 20.04 recommended an installation reference at the link below.  Skip the section of the guide about ROS if you do not need it.

https://github.com/IntelRealSense/realsense-ros/issues/1724#issuecomment-901465582

JHPHELAN, 2022-08-10:
> IntelRealSense/realsense-ros#1724 (comment)

It appears that link is deprecated:

"[deprecated] Feb 2021
Ubuntu 20.04 LTS is now natively supported by the RealSense™ SDK 2.0."

and suggests

"Have a look on https://github.com/IntelRealSense/lib..."

Have a look on https://github.com/IntelRealSense/lib...

J

James H Phelan
"Nihil est sine ratione cur potius sit quam non sit"
Leibniz

"Here am I, the servent of the Lord;
let it be with me, according to your Word"
Luke 1:38

MartyG-RealSense, 2022-08-10:
That statement did not include **Arm64 package** installation of the librealsense for Ubuntu 20.04 Focal.

Ecophagy, 2022-09-05:
Bump for any updates for an official apt package on Ubuntu 20.04 arm64? It would be tremendously helpful if any information could be given about timelines, since I would prefer to avoid compiling from source if possible.

MartyG-RealSense, 2022-09-05:
The latest information that I have now from my Intel RealSense colleagues is that Arm64 package support for Ubuntu 20.04 will not be added in the foreseeable future unfortunately and the SDK must therefore be compiled from source code for Arm64 on Ubuntu 20.

Ecophagy, 2022-09-08:
That's unfortunate, but thanks for your swift response.

I did find an alternative to compiling from source - for some reason  ROS maintains their own version of librealsense as a dependency for realsense2-camera which has been published for arm64 Ubunutu 20.04. It can be found on apt as ros-noetic-librealsense2. It seems to work fine, although I don't know for sure if it is complete or will be updated.

MartyG-RealSense, 2022-09-13:
Does anyone on this discussion require further assistance, please?  Thanks!

JHPHELAN, 2022-09-13:
If Intel is not going to support a package for the ARM64, then I guess not.  :-(

MartyG-RealSense, 2022-09-13:
Thanks very much @JHPHELAN for the update.  I apologize that the outcome was not the one that you hoped for.

closed 2022-09-19 by MartyG-RealSense