realsenseai/librealsense #9042: L515 Model's Depth Sensor issue (invalid value in rs2_open_multiple)

Hwoongg, 2021-05-18:
| | |
|---|---|
|**librealsense**|2.44.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0350937|
|**Firmware Version**|01.05.05.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b52476ff|
|**Firmware Update Id**|0003b52476ff|

When I start streaming the depth sensor I get this error.

Invalid Value in rs2_open_multiple(sensor:000002843434C550,
profiles:00000284333B48D8, count:2):
[image]

RealSenseSupport, 2021-06-15:
@Hwoongg 

Does this happen all the time?  Are you using any other USB3 connector other than the cable that came with the camera?  Can you attempt to change orientation of the USBC connection point and flip it and try?  Are you able to get RGB stream as well as IMU stream?  If this cannot be resolved via FW update, then it might be a hardware issue where you need to RMA.

You can also submit a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

RealSenseSupport, 2021-06-30:
@Hwoongg Do you need further help? If we don't hear from you in 7 days, this issue will be closed.

Hwoongg, 2021-07-02:
I tried changing the USBC connection direction but got the same result. RGB data can be obtained, but depth shooting is not activated. I bought another one of the same model and connected it to it, and it works fine. It doesn't seem to be a problem with the PC or the cable. It looks like the hardware needs repair. 
 

ㅇ
ㅇ

RealSenseSupport, 2021-07-06:
@Hwoongg Please do the RMA.

closed 2021-11-18 by RealSenseCustomerSupport