realsenseai/librealsense #9043: Does Jetson L4T 32.5.1 contain UVC header for D455?

MoBaT, 2021-05-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.9.201-tegra                                         | 
| Platform                                 | Jetson L4T 32.5.1  |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C++                          | 

### Issue Description
With the current L4T version I am using, I am trying to get the synchronized system timestamp for both my D435I sensor and D455. The problem is that when I try to get the timestamp for the D455, I get "UVC header is not available". For the D435I it works perfectly. I know with the L4T version I am on, they added the kernel driver to get UVC header to work. But does it only work for the D435I? I upgraded to this L4T version to avoid compiling the kernel myself so I can get synchronized timestamps.

MartyG-RealSense, 2021-05-19:
Hi @MoBaT   I note that you are using SDK version 2.38.1 according to the information provided above.  Support for L4T 32.5 (JetPack 4.5) was added in SDK version **2.43.0**.

[image]

If you update to SDK 2.43.0 and still experience UVC problems with your D455, you could consider rolling back from L4T 32.5.1.1 to **32.5** since 32.5 is the version that is stated to be officially supported.

MartyG-RealSense, 2021-05-19:
Thanks very much for the update.  I will await your test report.

MartyG-RealSense, 2021-05-19:
I am not involved in T265 support, so please do not include it in this particular discussion.  Thanks!

If your test does not work for you then I will seek advice from Intel about your problem.

MoBaT, 2021-05-19:
@MartyG-RealSense, good news! Running the `patch-realsense-ubuntu-L4T.sh` made it work. So for others who find this thread, the kernel headers in L4T 32.5 and L4T 32.5.1 are not sufficient to get the D455 to work. You need to run the patch script.

MartyG-RealSense, 2021-05-19:
Great news - thanks so much for running the tests and sharing the knowledge with the RealSense community!

closed 2021-05-19 by MoBaT