realsenseai/librealsense #9044: Issue installing librealsense on Focal in ARM

sisaha9, 2021-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | None yet| 
| Operating System & Version |  Ubuntu 20.04 Jetson  | 
| Kernel Version (Linux Only)    |  Not sure   | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |   2                    | 
| Language                            |  Any           | 
| Segment			|  Robot | 

### Issue Description
I was following this guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md. Since I am running it on a ROS2 Dockerfile on Jetson (arm64v8/ros:foxy) I updated it from bionic to focal. However I am unable to find any librealsense packages under it. It seems it has not been built for ARM yet. What are my alternatives?

MartyG-RealSense, 2021-05-19:
Hi @sisaha9   If you have problems with installing librealsense from packages on Jetson, an alternative is to build from source code.  The link below provides instructions from doing so with CMake on Jetson.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

sisaha9, 2021-05-19:
Hi @MartyG-RealSense thanks for the comment link! Would that work with Foxy? I believe cmake isn't called directly anymore as we use the colcon utility

MartyG-RealSense, 2021-05-19:
In regard to building the ROS2 Foxy wrapper on Focal with CMake, there have been reported problems with colcon when doing so, such as the case in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1725

**Doronhi** the RealSense ROS wrapper developer recommended avoiding setting **realsense2_DIR**

https://github.com/IntelRealSense/realsense-ros/issues/1399#issuecomment-695747807

MartyG-RealSense, 2021-05-25:
Hi @sisaha9  Do you require further assistance with this case, please?  Thanks!

sisaha9, 2021-05-25:
Oh sorry. We did try but our CMake kept giving an error with the Makefile even after disabling CUDA. This was in a Jetson Dockerfile with base arm64v8/ros:foxy

MartyG-RealSense, 2021-05-25:
The recent case in the link below about support for Arm64 packages on Focal would seem to be relevant to your situation.  Do you agree please?

https://github.com/IntelRealSense/librealsense/issues/9040#issuecomment-847442449

sisaha9, 2021-05-25:
Yes I agree

MartyG-RealSense, 2021-05-25:
Okay, thanks very much.  As mentioned in the other case, I have sought advice from Intel about this matter.

sisaha9, 2021-05-25:
Understood. I can close this issue then and follow that thread for any updates

closed 2021-05-25 by sisaha9