realsenseai/librealsense #9045: Issue of RGB color stream turning purple when auto exposure is enabled

victorll998, 2021-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | v2.45.0  | 
| Operating System & Version |  ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-50                                         | 
| Platform                                 | Intel Nuc  |
| SDK Version                            | 2.45.0                        | 
| Language                            | C/C++                       | 
| Segment			|  Other                        | 

### Issue Description
Hi there,

When I use D455 outside I found the RGB color stream went for being purplish like below. Does anyone have any idea about it? Or if there is any way to reverse it. Thank you.

[image]

FPSychotic, 2021-05-20:
[image]

This is my replacement,.the camera is unusable, just like the other one.
There is not filter or Firmware that will fix the issue, because add the filter add other issues in exposition, etc. It is just a faulty hardware due to some design , or quality of the UV filter. 
How you make a robot to track colors?, How work with AI? It is not RGB, It is RGB-IR, I can see the IR points in my RGB sensor, no only the sun, and is not with sun, it happens with is cloudy, in the shadow..., what design have the camera to add filters? This camera is not designed to add any filter, it is just faulty. What I explain to a customer that receive this camera if ask me? 
How I extract useful data from violet pointclouds?
Check this video , the color, frame speed, exposure time, the errors on exposure, in a cloudy day. no think in your application or your customer, and how you explain to him, the movie that intel invented, that the pixel size..that put sunglasses..that global shutter sensitivity...and now check that same rgb sensor in any other camera that use it, and you will see that have not this problem, because is not the sensor, is the filter that is faulty, bad designed,  supplier or intel fault in QC, but it is not fixeable, and of course is not a normal operation 
https://youtu.be/LzL1Vo-Bow8

FPSychotic, 2021-05-20:
> [image]
>
> This is my replacement,.the camera is unusable, just like the other one.
> There is not filter or Firmware that will fix the issue, because add the filter add other issues in exposition, etc. [...]

And again, why you closed the other issue?

RealSenseSupport, 2021-06-17:
“The Intel® RealSense™ Depth Camera D455 is used for longer range depth applications, with a wide field of view and global shutter on the depth and RGB sensors. The integrated RGB sensor over-saturates in bright sunlight and presents itself as a purple hue. We do not see this phenomenon indoors or outdoors without direct bright sunlight. This has no effect on depth quality. We are able to mitigate some of the effects with options shown in the “Specification Clarification” document at https://dev.intelrealsense.com/docs/firmware-releases. We are also working with our supplier and researching other ways to improve the saturation effect.”

MartyG-RealSense, 2026-10-25:
This open RealSense GitHub issue is being closed because of its significant age.  Thank you.

closed 2026-10-25 by MartyG-RealSense