realsenseai/librealsense #9047: What is the meaning of rotation data return from T265?

JimXu1989, 2021-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T 265 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 20.04)   | 
                                   | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<45>.<0> }                          | 
| Language                            |  {python }                          | 

### Issue Description
I use the python example provided offically,"t265_example.py".
The pose returned is as shown below:
[image]
the rotation data is return in 4 value:
x: 0.109211, y: -0.855083, z: 0.116369, w: -0.49332
I want to know the meaning of these value, and I want to know how to convert these values to a rotation matrix. As the matrix I can use in open3d which I can use to SLAM.

Thanks a lot!

sumitsarkar1, 2021-05-23:
these are rotations expressed in Unit Quaternions. Google it !

JimXu1989, 2021-05-24:
> these are rotations expressed in Unit Quaternions. Google it !

Thanks you

closed 2021-05-24 by JimXu1989