realsenseai/librealsense #9048: L515 sensor stuck in USB 2.1

SidNino, 2021-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  legacy / 2.45.0                           | 

### Issue Description
I have 2 Realsense L515 Lidar camera.For few days both the cameras were showing as usb 3.0 on realsense viewer. To work with ROS2 we updates the realsense viewwer to 2.45 from 2.41 and updated the firmware. We did not plug both of them together ever.Now one of them always shows up as usb 2.1 while the other one shows up as usb 3.0 . We have tried the following :

- Plugging the device in different windows PC 
- Plugging it into other identical Jetson TX2 with same information 
- Plugging the camera end first and then connecting the USB type A port 
- Connecting the Type C end of the wire both the ways
- Using the cable from other "working" sensor to check if its a wire issue . 

Typing `lsusb` in the terminal window gives the following output 

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8086:0b64 Intel Corp. 
Bus 001 Device 004: ID 1ea7:0064  
Bus 001 Device 003: ID 04d9:1702 Holtek Semiconductor, Inc. Keyboard LKS02
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Typing `lsusb -t` in the terminal window gives the following output 

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/3p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 7, If 0, Class=Video, Driver=, 480M
    |__ Port 2: Dev 7, If 7, Class=Human Interface Device, Driver=usbhid, 480M
    |__ Port 2: Dev 7, If 5, Class=Video, Driver=, 480M
    |__ Port 2: Dev 7, If 3, Class=Video, Driver=, 480M
    |__ Port 2: Dev 7, If 1, Class=Video, Driver=, 480M
    |__ Port 2: Dev 7, If 6, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 2: Dev 7, If 4, Class=Video, Driver=, 480M
    |__ Port 2: Dev 7, If 2, Class=Video, Driver=, 480M
```

Typing `rs-enumerate-devices` in the terminal window gives the following output 

```
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0264451
    Firmware Version              : 	01.05.02.00
    Recommended Firmware Version  : 	01.05.05.00
    Physical Port                 : 	1-2-7
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	L500
    Asic Serial Number            : 	0003ac5441a8
    Firmware Update Id            : 	0003ac5441a8

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence	  320x240	@ 30Hz	   RAW8
    Infrared	  320x240	@ 30Hz	   Y8
    Depth	  320x240	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
    Color	  640x480	@ 30Hz	   RGBA8
    Color	  640x480	@ 30Hz	   BGR8
    Color	  640x480	@ 30Hz	   YUYV
    Color	  640x480	@ 15Hz	   RGB8
    Color	  640x480	@ 15Hz	   Y16
    Color	  640x480	@ 15Hz	   BGRA8
    Color	  640x480	@ 15Hz	   RGBA8
    Color	  640x480	@ 15Hz	   BGR8
    Color	  640x480	@ 15Hz	   YUYV
    Color	  640x480	@ 6Hz	   RGB8
    Color	  640x480	@ 6Hz	   Y16
    Color	  640x480	@ 6Hz	   BGRA8
    Color	  640x480	@ 6Hz	   RGBA8
    Color	  640x480	@ 6Hz	   BGR8
    Color	  640x480	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F

```

Is there any solution to this problem ? i saw some similar issues on the github page but none of them seem to give me a definite solution .

RealSenseSupport, 2021-06-10:
@SidNino 

If you're having issues with the streaming of the depth in the manner outlined above or even having issues with the USBC connection orientation changing behavior with regards to detected as USB2 or USB3, please open a ticket here: https://support.intelrealsense.com/hc/en-us/requests/new

Did you submit the request at the link above?

SidNino, 2021-06-10:
Yes, I had already submitted an issue and they have arranged for a replacement now.

closed 2021-06-10 by SidNino