realsenseai/librealsense #9052: D435 SDK python setup for better depth output

pranav-khekare, 2021-05-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 and D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.45.0) | 
| Operating System & Version |   {Linux (Ubuntu 18.04.5) / MacOS  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy }                          | 
| Language                            |  {python}                          | 
| Segment			|                      | 

I am correctly able to run the python script explained in the examples provided. But the depth images that I get from running the script dont look nearly as good as when I take them from the viewer. I am also turning on the advanced mode and loading the config.json file from viewer to set the sensor to take the exact same depth image and still dont see the same results. Any help with setting up to get similar results? Also how to enable hole filling in SDK?

MartyG-RealSense, 2021-05-19:
Hi @pranav-khekare    The RealSense Viewer applies a range of colorizer settings and post-processing filters by default.  In scripts that you create yourself, these settings have to be deliberately programmed into the application, which means that such applications can look different to the images in the Viewer.

Python programming resources for Histogram Equalization, color scheme and color preset can be found in the links below.

https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

https://github.com/IntelRealSense/librealsense/issues/8877#issuecomment-825795803

In regard to hole-filling, you can do so by applying a post-processing **Hole-Filling Filter**.  Intel provide a Python post-processing tutorial in the link below that includes hole-filling.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

pranav-khekare, 2021-05-19:
Hey @MartyG-RealSense,
Thank you for the response. I will check these out, as long as the depth information is correct, thats what matters. Since it is difficult to visualize depth information, looking at the depth images from opencv visualization, it feels like the sdk collect different depth information from the viewer.

MartyG-RealSense, 2021-06-02:
Hi @pranav-khekare  Do you require further assistance with this case, please?  Thanks!

pranav-khekare, 2021-06-07:
Hey @MartyG-RealSense, sorry for the late response. What you mentioned worked out, Thank you for your help.

MartyG-RealSense, 2021-06-07:
Excellent news @pranav-khekare - thanks very much for the update!

closed 2021-06-13 by MartyG-RealSense