realsenseai/librealsense #9053: Mac OS Catalina fails to build librealsense from source

alifeinbinary, 2021-05-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.09.00) | 
| Operating System & Version |   {MacOS Catalina 10.15.7  | 
| Platform                                 | Mac OS  |
| SDK Version                            |  { 2.45.0 }                          | 
| Language                            |  {C/C#/opencv/python }                          | 
| Segment			|  {building from source }                          | 

### Issue Description
Following this guide [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md)
## Replicating the issue

```
git clone git@github.com:IntelRealSense/librealsense.git
cd librealsense/
mkdir build && cd build/
cmake .. -DBUILD_PYTHON_BINDINGS:bool=true -DOPENSSL_INCLUDE_DIR=/usr/bin/openssl -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DCMAKE_BUILD_TYPE=Release
make -j16
```
I have to specify the `OPENSSL_INCLUDE_DIR` variable or else cmake will throw an error saying it can't find openssl

I had librealsense installed via homebrew ```brew install librealsense``` but have since uninstalled it, no change.
I have re-installed libusb via homebrew
`pkg-config` and `vulkan-sdk` are installed via homebrew
I have set `/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`
I have tried running cmake with only ```-DOPENSSL_INCLUDE_DIR=/usr/bin/openssl``` specified, no change. 

Every time I run `make -j16` I get this error. 
```
/Users/andrew/projects/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning:
      token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_DEBUG, fmt, ##__VA_ARGS__)
                                                                                         ^
/Users/andrew/projects/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning:
      token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
[ 49%] Building CXX object CMakeFiles/realsense2.dir/src/types.cpp.o
[ 49%] Building C object CMakeFiles/realsense2.dir/src/verify.c.o
[ 50%] Building CXX object CMakeFiles/realsense2.dir/src/depth-to-rgb-calibration.cpp.o
[ 50%] Building C object CMakeFiles/realsense2.dir/third-party/sqlite/sqlite3.c.o
[ 50%] Building CXX object CMakeFiles/realsense2.dir/third-party/easyloggingpp/src/easylogging++.cc.o
/Users/andrew/projects/librealsense/third-party/sqlite/sqlite3.c:20294:17: warning:
      'OSAtomicCompareAndSwapPtrBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      atomic_compare_exchange_strong() from <stdatomic.h> instead [-Wdeprecated-declarations]
      success = OSAtomicCompareAndSwapPtrBarrier(NULL, newzone,
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/usr/include/libkern/OSAtomicDeprecated.h:548:6: note:
      'OSAtomicCompareAndSwapPtrBarrier' has been explicitly marked deprecated here
bool    OSAtomicCompareAndSwapPtrBarrier( void *__oldValue, void *__newValue, void * volatile *__theValue );
        ^
[ 50%] Building CXX object CMakeFiles/realsense2.dir/common/utilities/time/work_week.cpp.o
[ 51%] Building CXX object CMakeFiles/realsense2.dir/common/utilities/time/l500/get-mfr-ww.cpp.o
2 warnings generated.
2 warnings generated.
1 warning generated.
[ 51%] Linking CXX shared library librealsense2.dylib
[ 51%] Built target realsense2
make: *** [all] Error 2
```

This isn't the first time I've built librealsense from source on this machine, it has worked in the past. I was trying to update my software, specifically the python package, as there's no pypi wheel for Mac OS.

MartyG-RealSense, 2021-05-20:
Hi @alifeinbinary   May I first ask whether you are using a recent Mac model with an **Apple Silicon M1** processor chip, please?  Thanks!

alifeinbinary, 2021-05-20:
@MartyG-RealSense No M1. I'm using a MacPro7,1 Intel Xeon W 3.2 Ghz 16-core.

MartyG-RealSense, 2021-05-20:
Thank you.  If the same build method has worked for you in the past on the same computer then you could try **make -j8** instead of j16 if you have not done so already, in order to eliminate the possibility that the build is failing because of a high j-number (I know that you have a powerful 16-core processor though).

alifeinbinary, 2021-05-20:
@MartyG-RealSense I failed to mention that I did try `make -j8`. I'll try it again with different build variable combinations.

MartyG-RealSense, 2021-05-20:
Another Mac user was recently having problems with a missing **OPENSSL_INCLUDE_DIR** when building librealsense.  Another user shared a fix that worked for them.

https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782

alifeinbinary, 2021-05-20:
Thanks @MartyG-RealSense. This thread got me to a completed build. 

Investigating a little further...
`which openssl` on my system revealed that MacOS defaults to using openssl at `/usr/bin/openssl`, which is actually LibreSSL (version 2.8.3 in my case). 

Using `brew unlink openssl && brew link openssl --force` resulted in a message instructing my to export the following variables in order for openssl, not LibreSSL, to be used during builds. It was this particular detail that got me to ta completed build using `cmake .. -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false`  
```
export PKG_CONFIG_PATH="/usr/local/opt/openssl@1.1/lib/pkgconfig"
export CPPFLAGS="-I/usr/local/opt/openssl@1.1/include"
export LDFLAGS="-L/usr/local/opt/openssl@1.1/lib"
```

closed 2021-05-20 by alifeinbinary

MartyG-RealSense, 2021-05-20:
Great to hear that you were successful @alifeinbinary - thanks very much for sharing the details of your investigation with the RealSense community!