realsenseai/librealsense #9054: How to set max distance when recording on D455

Jonapple, 2021-05-20:
| Camera Model | { D455 } |
| Firmware Version | 05.12.10.00 |
| Operating System & Version | MacOS |
| Platform | Mac|
| SDK Version | 2.2.42.0 |
| Language | C++ |
| Segment | Ply, RGBD, Recording |

Hi,

I'm wondering if it is possible to set the max distance in the realsense-viewer when recording.

I've already set the max distance to a value I want
[image]

But after I extra depth information from the bag file (i.e: a ply file), the object out of that distance is still recorded. Below is an example:

RGB image: The toy was about within 0.5m from the camera, and the wall has at least 1.5M away
[image]
Visualise the Ply: everything behind 1M was still included
[image]
Here is a scaled depth image extract from the bag file
[image]

I wish only to record the toy, i.e: within 1M, and ignore everything out of that distance. Is there any setting in the realsense-viewer to achieve this?

Cheers.

MartyG-RealSense, 2021-05-20:
Hi @Jonapple   The **Histogram Equalisation** option alters how color is rendered.  To set depth rendering distance limits you should use the **Threshold Filter** post-processing filter that can be found under the **Post-Processing** category of the Stereo Module options.  The Threshold Filter is set to a maximum distance of 4 meters by default.

It should be noted though that a bag file is a recording of raw, unprocessed camera data.  Once a bag file is loaded into memory, then post-processing can be performed on the data in **real-time**, such as changing the maximum distance of depth information that is displayed.  You can then go into **3D** point cloud mode to export a ply file.

MartyG-RealSense, 2021-05-31:
Case closed due to no further comments received.

closed 2021-05-31 by MartyG-RealSense

Jonapple, 2021-05-31:
> Hi @Jonapple Do you require further assistance with this case, please? Thanks!

Sorry for the late reply, the problem is sloved. Thank you for mention the bag file could be reloaded and post-process, saved my life in data collecting.

MartyG-RealSense, 2021-05-31:
Great news that you were successful @Jonapple - thanks very much for the update!