realsenseai/librealsense #9056: How to get the point cloud of each frame of a video through matlab？

rain0902, 2021-05-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10  |                                      | 
| SDK Version                            |  { legacy / 2.43 }                          | 
| Language                            |  matlab }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I used realsense D435 to shoot a video about the surface deformation of a soft object, and I want to get the point cloud of each frame of the video. 

But pointcloud_example seems to only get the pointcloud of one frame , how to get the point cloud of each frame of a video through matlab？Please.

MartyG-RealSense, 2021-05-20:
Hi @rain0902  Is the video that you recorded a **bag** file, please?

If it is a bag, the subject of generating a pointcloud in the RealSense MATLAB wrapper from all frames of a bag has been previously discussed, though it is a complicated subject.  There is a long and detailed case about it in the link below.

https://github.com/IntelRealSense/librealsense/issues/8368

rain0902, 2021-05-21:
hello, thank you very much for your reply.

I have read the content in the link , https://github.com/IntelRealSense/librealsense/issues/8368
but it is too confusing and complicated that I don’t understand it.

Can you give me a clear and concise explanation？Please!

Thank you in advance

MartyG-RealSense, 2021-05-21:
I am not a RealSense MATLAB programmer so I am not the best person to explain the content of the link, unfortunately.

The link below provides details of further information resources for using RealSense with MATLAB that you could study.

https://github.com/IntelRealSense/librealsense/issues/8241

MartyG-RealSense, 2021-05-27:
Hi @rain0902   Were you able to resolve your MATLAB problem, please?  Thanks!

MartyG-RealSense, 2021-06-07:
Case closed due to no further comments received.

closed 2021-06-07 by MartyG-RealSense

rain0902, 2021-06-07:
Yes, I can't solve this question.It's too difficult!

MartyG-RealSense, 2021-06-07:
I researched your question again.  Whilst I did not find a pointcloud solution, I did locate a RealSense MATLAB project that can generate a live video image like the one in the **2D** mode of the RealSense Viewer.  If you can get the script provided in this project working then it may act as a useful reference for then generating a point cloud.

https://www.instructables.com/Final-Project-for-Introduction-to-Robotics-Using-a/

In **Step 4** of the guide is a text file called **livetracking text.txt**.  If you open this file, inside is a RealSense MATLAB wrapper script that you can use to create a **.mat** file.

[image]

I have pasted the script below in case you cannot open the txt file on your computer.

***********************

```
function test_somethinga();
  
    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    % Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();
       imgstore=[];
    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();
figure('visible','on');  hold on;
    figure('units','normalized','outerposition',[0 0 1 1])
    % Get streaming device's name
    dev = profile.get_device();
    name = dev.get_info(realsense.camera_info.name);

    % Get frames. We discard the first couple to allow
    % the camera time to settle
    
    for i = 1:200
        
        fs = pipe.wait_for_frames();
            % Stop streaming

    % Select depth frame
    depth = fs.get_depth_frame();
    % Colorize depth frame
    color = colorizer.colorize(depth);

    % Get actual data and convert into a format imshow can use
    % (Color data arrives as [R, G, B, R, G, B, ...] vector)
    data = color.get_data();
    img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
    img(:,1:100,:)=[];
    img(:,840:end,:)=[];
    img(460:480,:,:)=[];
    A(:,:,:)=img(:,:,:);
    % Display image
        g=rgb2gray(img);

size(g);

level=graythresh(g);

BW2 = edge(g,'Prewitt');

   Ibw1 = imfill(BW2,'holes');
binaryImage = bwareafilt(Ibw1, 1); % Extract largest blob only.

cleanedBinaryImage = bwareaopen(binaryImage, 15); % Remove blobs smaller than 40.

info = regionprops(cleanedBinaryImage,'Boundingbox');
for k = 1 : length(info)
     BB = info(k).BoundingBox;
     X1(:,k)=[BB(1)];
     X2(:,k)=[BB(2)];
     X3(:,k)=[BB(3)+4];
     X4(:,k)=[BB(4)+4];
     X=transpose([X1;X2;X3;X4]);

end

IExpandedBBoxes = insertShape(img,'Rectangle',X,'LineWidth',3);

overlapRatio = bboxOverlapRatio(X, X);
n = size(overlapRatio,1); 
overlapRatio(1:n+1:n^2) = 0;
g1 = graph(overlapRatio);
componentIndices = conncomp(g1);
c=1;
% Merge the boxes based on the minimum and maximum dimensions.
xmin = accumarray(componentIndices', X1, [], @min);
ymin = accumarray(componentIndices', X2, [], @min);
xmax = accumarray(componentIndices', X3, [], @max);
ymax = accumarray(componentIndices', X4, [], @max);

textBBoxes = [xmin ymin xmax ymax];

ITextRegion = insertShape(img, 'Rectangle', textBBoxes,'LineWidth',3);
 s(i,:)= [regionprops(ITextRegion,'centroid')];
xcenter=round(xmin +(xmax/2));
ycenter=round(ymin+(ymax/2));
cent=g(xcenter,ycenter);

str = strcat('<HI,1,2,',num2str(cent),',',num2str(xcenter),',',num2str(ycenter),'>')

imshow(ITextRegion)

    end
        pipe.stop();
     
end
```

rain0902, 2021-06-07:
Thank you very very much! This is so good

MartyG-RealSense, 2021-06-07:
You are very welcome @rain0902    :)