realsenseai/librealsense #9059: Colored lines at the edge of depth image in Realsense Viewer

Marenix, 2021-05-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.12.13.50 | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.4.0-73                                        | 
| Platform                                 | PC |
| SDK Version                            | 2.45.0                      | 
| Language                            |                       | 
| Segment			|                         | 

### Issue Description
Hey,
I've been testing D415 and noticed some lines at the edges of the depth image. They appear to show something there, even though there''s nothing, and when I zoom in and hover over that colored area, it shows the depth is 0m. It also appears to change based on the rest of the image, like it's showing maximum/minimum in a row/column. I tried some of the older versions of librealsense, and I found that this appears from version 2.24.0. Versions before that don't have it, and the release notes for version 2.24 didn't have anything about it, so I'm guessing it's not intentional?

It's seen most clearly at the left edge, but it's also visible at the upper and lower edge:
[image]

The zoomed in version, with my pointer hovering over the dark red area at the edge:
[image]

MartyG-RealSense, 2021-05-20:
Hi @Marenix  I have handled a few past cases involving this phenomenon, though the colored line was limited to the bottom of the image rather than being present on the side too.

https://github.com/IntelRealSense/librealsense/issues/6845
https://github.com/IntelRealSense/librealsense/issues/7340
https://support.intelrealsense.com/hc/en-us/community/posts/360048022393-Dead-green-pixels-in-stream-

My understanding is that it occurs on Linux but not on Windows, and that it can be resolved by building librealsense from source code using the **RSUSB backend** installation method.  This method, which uses an internet connection, is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

Marenix, 2021-05-20:
Hi @MartyG-RealSense, thanks for the quick answer! You are correct, I tested it on Windows and this phenomenon isn't present there. I have also tried your method, building librealsense from source using RSUSB backend on Linux, and unfortunately it did not solve the problem. The phenomenon is still there. From what I saw in the links you shared, other people had this problem on RGB image, while here it appears on depth image.

MartyG-RealSense, 2021-05-21:
Hi @Marenix  After discussion with Intel, I have filed an official bug report with Intel regarding the issue of colored lines on edges on Linux only.  This case should be kept open whilst the bug report is active.  Thanks for the report!

Marenix, 2021-05-24:
Hi @MartyG-RealSense, thanks for the info!

MartyG-RealSense, 2023-01-24:
Hi @Marenix  My Intel RealSense colleagues have now implemented a fix in the SDK for the issue of colored lines on the edges of the depth image.  Thanks very much for your patience!  As the solution is implemented, this case will be closed.  Thanks again!

closed 2023-01-24 by MartyG-RealSense

MartyG-RealSense, 2023-01-25:
The change has been merged into the **development** version of the SDK, which becomes the latest 'master' SDK version when released.  So it will be in the next release, which should be **2.54.0**.

The development branch of the SDK can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/development