realsenseai/librealsense #9062: dmesg messages issue

Allius27, 2021-05-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.4.0-81-generic  |
| Platform                                 | PC |
| SDK Version                            |  2.39  |
| Language                            |  C++  |
| Segment			|  others |

Hi there. 
We build realsense from source.
We don't see any problems in camera work, but from time to time kernel catches that messages

What that errors might mean?
Happens on starting application

<details>
  <summary>$ dmesg -w</summary>

```
[102910.434925] usb 1-2: USB disconnect, device number 16
[102911.242424] usb 1-2: new high-speed USB device number 17 using xhci_hcd
[102911.371463] usb 1-2: New USB device found, idVendor=8086, idProduct=0b07
[102911.371474] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[102911.371479] usb 1-2: Product: Intel(R) RealSense(TM) Depth Camera 435
[102911.371483] usb 1-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
[102911.373346] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[102911.373359] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[102911.373665] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[102911.376235] uvcvideo: Unable to create debugfs 1-17 directory.
[102911.376806] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/input/input15
[102911.377757] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[102911.378891] uvcvideo: Unable to create debugfs 1-17 directory.
[102912.059425] ------------[ cut here ]------------
[102912.059450] WARNING: CPU: 1 PID: 737 at /build/linux-cs3yMe/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[102912.059453] Unknown pixelformat 0x2036315a
[102912.059455] Modules linked in: ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack x_tables nf_nat nf_conntrack br_netfilter bridge stp llc aufs overlay snd_hda_codec_hdmi intel_rapl snd_hda_codec_realtek snd_hda_codec_generic intel_powerclamp coretemp kvm_intel btusb btrtl uvcvideo videobuf2_vmalloc videobuf2_memops btbcm btintel videobuf2_v4l2 kvm videobuf2_core v4l2_common bluetooth videodev rtsx_pci_ms snd_intel_sst_acpi input_leds irqbypass serio_raw snd_hda_intel snd_intel_sst_core memstick media punit_atom_debug snd_soc_rt5670 snd_hda_codec snd_soc_rl6231 snd_soc_sst_mfld_platform snd_hda_core snd_soc_core nxp_nci_i2c snd_compress ac97_bus dw_dmac
[102912.059518]  dw_dmac_core nxp_nci shpchp snd_pcm_dmaengine snd_hwdep nci snd_pcm mei_txe nfc dwc3 udc_core mei i2c_designware_platform i2c_designware_core rfkill_gpio snd_timer acpi_pad snd 8250_dw ulpi soundcore spi_pxa2xx_platform lpc_ich soc_button_array pwm_lpss_platform mac_hid pwm_lpss ib_iser rdma_cm iw_cm ib_cm ib_sa ib_mad ib_core ib_addr iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi autofs4 btrfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid uas usb_storage rtsx_pci_sdmmc crct10dif_pclmul crc32_pclmul ghash_clmulni_intel i915 aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd i2c_algo_bit drm_kms_helper psmouse syscopyarea sysfillrect sysimgblt r8169 fb_sys_fops mii
[102912.059594]  rtsx_pci drm ahci libahci fjes video i2c_hid hid sdhci_acpi sdhci
[102912.059607] CPU: 1 PID: 737 Comm: atm-rsserver Tainted: G        W       4.4.0-81-generic #104-Ubuntu
[102912.059610] Hardware name: GIGABYTE GB-BACE-3160/MZBSWMP-00, BIOS F6 06/13/2018
[102912.059613]  0000000000000286 00000000ca00aeb7 ffff8800051ebbf0 ffffffff813f9683
[102912.059618]  ffff8800051ebc38 ffffffffc079da78 ffff8800051ebc28 ffffffff81081322
[102912.059623]  ffff8800051ebda0 0000000000000000 0000000000000000 ffff880034941c00
[102912.059627] Call Trace:
[102912.059637]  [<ffffffff813f9683>] dump_stack+0x63/0x90
[102912.059644]  [<ffffffff81081322>] warn_slowpath_common+0x82/0xc0
[102912.059647]  [<ffffffff810813bc>] warn_slowpath_fmt+0x5c/0x80
[102912.059655]  [<ffffffffc07d348d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[102912.059667]  [<ffffffffc07886f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[102912.059679]  [<ffffffffc07892c1>] __video_do_ioctl+0x291/0x310 [videodev]
[102912.059684]  [<ffffffff812199e4>] ? terminate_walk+0x64/0xd0
[102912.059689]  [<ffffffff811ed199>] ? ___slab_alloc+0x1e9/0x470
[102912.059701]  [<ffffffffc0788d96>] video_usercopy+0x336/0x5b0 [videodev]
[102912.059712]  [<ffffffffc0789030>] ? video_ioctl2+0x20/0x20 [videodev]
[102912.059717]  [<ffffffff811ed9ea>] ? kmem_cache_alloc+0x1ca/0x1f0
[102912.059729]  [<ffffffffc0789025>] video_ioctl2+0x15/0x20 [videodev]
[102912.059740]  [<ffffffffc07846a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[102912.059744]  [<ffffffff8122320f>] do_vfs_ioctl+0x29f/0x490
[102912.059748]  [<ffffffff81223479>] SyS_ioctl+0x79/0x90
[102912.059754]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[102912.059757] ---[ end trace 02277c9d1f0b1a15 ]---
[102912.060067] ------------[ cut here ]------------
[102912.060083] WARNING: CPU: 1 PID: 737 at /build/linux-cs3yMe/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[102912.060086] Unknown pixelformat 0x00000000
[102912.060088] Modules linked in: ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack x_tables nf_nat nf_conntrack br_netfilter bridge stp llc aufs overlay snd_hda_codec_hdmi intel_rapl snd_hda_codec_realtek snd_hda_codec_generic intel_powerclamp coretemp kvm_intel btusb btrtl uvcvideo videobuf2_vmalloc videobuf2_memops btbcm btintel videobuf2_v4l2 kvm videobuf2_core v4l2_common bluetooth videodev rtsx_pci_ms snd_intel_sst_acpi input_leds irqbypass serio_raw snd_hda_intel snd_intel_sst_core memstick media punit_atom_debug snd_soc_rt5670 snd_hda_codec snd_soc_rl6231 snd_soc_sst_mfld_platform snd_hda_core snd_soc_core nxp_nci_i2c snd_compress ac97_bus dw_dmac
[... 31 lines cut ...]
[102912.060751] CPU: 1 PID: 737 Comm: atm-rsserver Tainted: G        W       4.4.0-81-generic #104-Ubuntu
[102912.060754] Hardware name: GIGABYTE GB-BACE-3160/MZBSWMP-00, BIOS F6 06/13/2018
[102912.060756]  0000000000000286 00000000ca00aeb7 ffff8800051ebbf0 ffffffff813f9683
[102912.060761]  ffff8800051ebc38 ffffffffc079da78 ffff8800051ebc28 ffffffff81081322
[102912.060765]  ffff8800051ebda0 0000000000000000 0000000000000000 ffff8800349419c0
[102912.060770] Call Trace:
[102912.060774]  [<ffffffff813f9683>] dump_stack+0x63/0x90
[102912.060779]  [<ffffffff81081322>] warn_slowpath_common+0x82/0xc0
[102912.060783]  [<ffffffff810813bc>] warn_slowpath_fmt+0x5c/0x80
[102912.060789]  [<ffffffffc07d348d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[102912.060801]  [<ffffffffc07886f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[102912.060812]  [<ffffffffc07892c1>] __video_do_ioctl+0x291/0x310 [videodev]
[102912.060817]  [<ffffffff812199e4>] ? terminate_walk+0x64/0xd0
[102912.060821]  [<ffffffff811ed199>] ? ___slab_alloc+0x1e9/0x470
[102912.060833]  [<ffffffffc0788d96>] video_usercopy+0x336/0x5b0 [videodev]
[102912.060844]  [<ffffffffc0789030>] ? video_ioctl2+0x20/0x20 [videodev]
[102912.060849]  [<ffffffff811ed9ea>] ? kmem_cache_alloc+0x1ca/0x1f0
[102912.060860]  [<ffffffffc0789025>] video_ioctl2+0x15/0x20 [videodev]
[102912.060871]  [<ffffffffc07846a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[102912.060875]  [<ffffffff8122320f>] do_vfs_ioctl+0x29f/0x490
[102912.060879]  [<ffffffff81223479>] SyS_ioctl+0x79/0x90
[102912.060884]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[102912.060886] ---[ end trace 02277c9d1f0b1a17 ]---
```
</details>

MartyG-RealSense, 2021-05-21:
Hi @Allius27  My research indicates that the set of messages in your dmesg log may occur due to an unpatched kernel, perhaps because the patch application process was unsuccessful.   In the link below, a RealSense team member offers advice on verifying if patching has been successful.

https://github.com/IntelRealSense/librealsense/issues/6837

MartyG-RealSense, 2021-05-27:
Hi @Allius27  Do you require further assistance with this case, please?  Thanks!

Allius27, 2021-05-31:
Thank for your advice. 
Yes, the update process is unsuccessful.

closed 2021-05-31 by Allius27