realsenseai/librealsense #9066: rs2::config Match Streams' Resolutions

foohyfooh, 2021-05-20:
| Required Info                         |                                                     |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D400                                           | 
| Firmware Version                  | 05.11.01.100                               | 
| Operating System & Version |   Ubuntu 18.04                            | 
| Kernel Version (Linux Only)   |  5.4.0-72-generic                       | 
| Platform                                 | PC                                             |
| SDK Version                          |  2.44.0                                       | 
| Language                              |  C++   + OpenCV                      | 
| Segment			        | Robot                                       | 

### Feature Request Description

I am looking to use `cv::hconcat` and `cv::vconcat` to combine unaligned images from the camera, but currently the only way that I can get guarantee that the dimensions of the output images is to use the below code where I explicitly put the dimensions to match even though [rs-enumerate-devices output](https://github.com/IntelRealSense/librealsense/files/6518432/rs-enumerate-devices-output.txt) shows that they have matching resolutions with the same fps.
```
rs2::config config;
config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16);
config.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8);
```

Can there be methods added to `rs2::config` that matched the enabled streams to the highest or lowest matching resolutions available?

MartyG-RealSense, 2021-05-21:
Hi @foohyfooh  Including the RGB format in such a mechanism could be problematic, as it has minimum and maximum resolutions (highlighted in the table below) that are not supported on the depth and infrared formats.  So forcing depth and infrared to match to RGB resolutions could cause a **Failed to resolve requests** error at pipeline start as the program would be unable to fulfill the requested configuration.

It may be viable if color resolution was synced to depth instead, so that the minimum and maximum RGB resolutions possible were 424x240 and 1280x720.  There would also be the issue of setting an FPS speed that was supported by all formats though to ensure that they were all configured to operate at the same speed.

There are no plans to add such a feature though as far as I am aware.  Using the camera's default configuration or defining a specific **cfg** configuration for streams individually works well for the majority of RealSense users.

foohyfooh, 2021-05-21:
Thanks for your response.
While you have stated that there are no specific plans to add this feature, I want to still at least clarify my request. It was not to force the depth camera to go to a higher or lower resolution that it doesn't support, but rather to match the highest or lowest resolution that they both support, example being `Depth 1280x720 @ 6Hz Z16` and `Color 1280x720 @ 6Hz BGR8` are both supported and thus can be returned as a valid configuration of highest matching resolutions.

MartyG-RealSense, 2021-05-21:
Thank you very much for the clarification.

On the D435 / D435i and D455 camera models, the optimal depth accuracy resolution is **848x480**, whilst on D415 the optimal depth accuracy resolution is **1280x720**.  

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-start-by-operating-at-optimal-depth-resolution-for-the-camera

So whilst depth can be set to 1280x720, it may not be the best choice in terms of depth accuracy on the D435/i and D455 models.

MartyG-RealSense, 2021-05-27:
Hi @foohyfooh  Do you require further assistance with this case, please?  Thanks!

foohyfooh, 2021-05-27:
No further assistance is required.

MartyG-RealSense, 2021-05-27:
Okay, thanks very much @foohyfooh for the confirmation!  I will close the case now,

closed 2021-05-27 by MartyG-RealSense

foohyfooh, 2021-05-27:
I have made a naive implementation that addresses the feature request for anyone who comes along with a similar issue.

```C++
#include <set>
#include <tuple>
#include <iostream>

using resolution_tuple = std::tuple<int, int, int>;

std::set<resolution_tuple> getResolutionIntersection(rs2_stream color, rs2_format color_format, rs2_stream depth, rs2_format depth_format) {
    // TODO: Make the device a parameter
    // TODO: Extend for more than 2 sensors
    rs2::context context;
    rs2::device_list devices_list = context.query_devices();
    rs2::device device = devices_list[0];
    std::vector<rs2::sensor> sensors = device.query_sensors();

    std::set<resolution_tuple> color_profiles, depth_profiles, intersection;

    for (rs2::sensor sensor: sensors){
        std::vector<rs2::stream_profile> profiles = sensor.get_stream_profiles();
        for(rs2::stream_profile profile: profiles){
            if (rs2::video_stream_profile video_profile = profile.as<rs2::video_stream_profile>()){
                resolution_tuple p = std::make_tuple(video_profile.width(), video_profile.height(), profile.fps());
                if(profile.stream_type() == color && profile.format() == color_format){
                    color_profiles.insert(p);
                } else if(profile.stream_type() == depth && profile.format() == depth_format){
                    depth_profiles.insert(p);
                }
            }
        }
    }

    std::set_intersection(color_profiles.begin(), color_profiles.end(), depth_profiles.begin(), depth_profiles.end(), std::inserter(intersection, intersection.begin()));
    return intersection;
}

rs2::config getLowestMatching(rs2_stream color, rs2_format color_format, rs2_stream depth, rs2_format depth_format) {
    std::set<resolution_tuple> intersection = getResolutionIntersection(color, color_format, depth, depth_format);
    // Since the set is internally sorted we can use intersection.begin() or (--intersection.rend()) to get the first element
    resolution_tuple resolution = *(intersection.begin());
    int width = std::get<0>(resolution), height = std::get<1>(resolution), fps = std::get<2>(resolution);
    std::cout << "width: " << width << " height: " << height << " fps: " << fps << std::endl;
    rs2::config config;
    config.enable_stream(color, width, height, color_format, fps);
    config.enable_stream(depth, width, height, depth_format, fps);
    return config;
}

rs2::config getHighestMatching(rs2_stream color, rs2_format color_format, rs2_stream depth, rs2_format depth_format) {
    std::set<resolution_tuple> intersection = getResolutionIntersection(color, color_format, depth, depth_format);
    // Since the set is internally sorted we can use intersection.rbegin() or (--intersection.end()) to get the last element
    resolution_tuple resolution = *(--intersection.end());
    int width = std::get<0>(resolution), height = std::get<1>(resolution), fps = std::get<2>(resolution);
    std::cout << "width: " << width << " height: " << height << " fps: " << fps << std::endl;
    rs2::config config;
    config.enable_stream(color, width, height, color_format, fps);
    config.enable_stream(depth, width, height, depth_format, fps);
    return config;
}
```

MartyG-RealSense, 2021-05-27:
Thanks so much @foohyfooh for sharing your code with the RealSense community!  :)