realsenseai/librealsense #9067: Error in depth measurement for L515

nikzadb, 2021-05-20:
Hi,

I use L515 to measure the length of an object sitting at 50cm from the sensor. The length of the object can be something between 0.5cm to 3.5cm. Looking at the documentation, it mentioned the AVG error is 5mm in 1m or 2.5mm in 50cm. Having a 2.5mm (0.25cm) error to measure the object will add a huge error on the object measurement:
- 0.5cm object: 50% error
- 3.5cm object: 7% error

How can I reduce this error? one way is to sample the object multiple time and do the average but I don't have proper access to the hardware (however, I can ask for this tweak).

Regards,
    Nikzad

zeynepkoyun, 2021-05-21:
Hi @nikzadb @RealSense-Customer-Engineering 

[image]
[image]
There are too many outliers because there is scattering in the point cloud information from the L515 camera.

Have you tried any method to extract outliers from the current point cloud?

nikzadb, 2021-05-21:
Thanks @zeynepkoyun 
I forgot to mention the Depth and RGB are aligned so both have the same size.
I didn't see outliers in my data so i can fairly assume it has been removed by sensor.
The issue, however, is that i expect pixels of an objest in the same physical distance of sensor have similar depths but due to the measurment error do not.
2.5 mm error in distance may be negligible in 50cm distance but is significant for 1cm object.

nikzadb, 2021-05-23:
Hi @MartyG-RealSense,

Does the SDK have the ability to take multiple measurements of the depth sensor and average it?

RealSenseSupport, 2021-06-10:
@nikzadb 

You'll have to handle averaging depth data over time within your code.  The SDK does not do the averaging inherently.

RealSenseSupport, 2021-06-18:
Will you be needing further help with this?
If we don’t hear from you in 7 days, this issue will be closed.

JZhu1208, 2021-07-06:
> Thanks @zeynepkoyun
> I forgot to mention the Depth and RGB are aligned so both have the same size.
> I didn't see outliers in my data so i can fairly assume it has been removed by sensor.
> The issue, [...]

Hello, 

How do you judge that "Depth and RGB are aligned"? And how to save the aligned image?

MartyG-RealSense, 2021-09-11:
Hi @nikzadb and @JZhu1208  Do either of you require further assistance with this case, please?  Thanks!

closed 2021-09-17 by MartyG-RealSense