realsenseai/librealsense #9068: Nodejs wrapper failing to build or install via npm on Mac OS Catalina

alifeinbinary, 2021-05-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.09.00) | 
| Operating System & Version |   {MacOS Catalina 10.15.7  | 
| Platform                                 | Mac OS  |
| SDK Version                            |  { 2.45.0 }                          | 
| Language                            |  {nodejs }                          | 
| Segment			|  {building nodejs from source }                          | 

### Issue Description
I'm trying to get the nodejs wrapper on my system but neither intsalling from npm or building from source is working for me. I'll post the print out from both, respectively.  

I have installed { libusb, pkg-config, glfw, cmake } using homebrew. 
I have jsdoc globally installed using npm. 
librealsense release is already built and installed on the system. 

Running ` npm install --save node-librealsense` from within my project virtualenv seems to fail specifically at:
```
npm ERR! ../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
npm ERR!     return v8obj_->Get(Nan::New(name).ToLocalChecked());
```
Have a look at the full print out
```
npm ERR! code 1
npm ERR! path /Users/andrew/projects/eel/node_modules/node-librealsense
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/node_librealsense/src/addon.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.2.0 | darwin | x64
npm ERR! gyp info find Python using Python version 3.9.5 found at "/Users/andrew/projects/eel/bin/python3"
npm ERR! (node:50808) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)
npm ERR! gyp info spawn /Users/andrew/projects/eel/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/andrew/projects/eel/node_modules/node-librealsense/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/andrew/Library/Caches/node-gyp/16.2.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/andrew/Library/Caches/node-gyp/16.2.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/andrew/Library/Caches/node-gyp/16.2.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/andrew/projects/eel/node_modules/node-librealsense',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
npm ERR!     return v8obj_->Get(Nan::New(name).ToLocalChecked());
npm ERR!            ~~~~~~~~^~~
npm ERR! /Users/andrew/Library/Caches/node-gyp/16.2.0/include/node/v8.h:3973:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
npm ERR!                                           ^
npm ERR! /Users/andrew/Library/Caches/node-gyp/16.2.0/include/node/v8.h:3976:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
[... 196 lines cut ...]
```
clang: error: no such file or directory: '/Users/andrew/projects/librealsense/wrappers/nodejs/../../build/Release/librealsense2.dylib'
make[3]: *** [Release/node_librealsense.node] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:365:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
gyp ERR! System Darwin 19.6.0
gyp ERR! command "/usr/local/Cellar/node/16.2.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/andrew/projects/librealsense/wrappers/nodejs
gyp ERR! node -v v16.2.0
gyp ERR! node-gyp -v v7.1.2
gyp ERR! not ok
npm ERR! code 1
npm ERR! path /Users/andrew/projects/librealsense/wrappers/nodejs
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(SOCKET, const char *, socklen_t, int)
-- Performing Test curl_cv_func_send_test

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/andrew/.npm/_logs/2021-05-21T03_44_45_862Z-debug.log
```

MartyG-RealSense, 2021-05-21:
Hi @alifeinbinary   It looks as though you are using Node.js version **16.2**.  The librealsense SDK currently supports up to Node.js version **14**.

[image]

alifeinbinary, 2021-05-21:
Good call @MartyG-RealSense, sorry I missed that! I reverted back to node v14.0.0, ran `npm install && npm install --save node-librealsense`. The build actually initiated this time and looked like it was about to finish, however, it stopped again at the same point. 

It looks like a dependency isn't getting included in `./wrappers/nodejs/src/addon.cpp`. Get,Set, Delete functions aren't linked?

```
> node-librealsense@0.2130.0 install /Users/andrew/projects/humanyse/eel/node_modules/node-librealsense
> node-gyp rebuild

  CXX(target) Release/obj.target/node_librealsense/src/addon.o
../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
    return v8obj_->Get(Nan::New(name).ToLocalChecked());
           ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3686:43: note: candidate function not viable: requires 2
      arguments, but 1 was provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                          ^
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3689:43: note: candidate function not viable: requires 2
      arguments, but 1 was provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                          ^
../src/addon.cpp:38:13: error: no matching member function for call to 'Set'
    v8obj_->Set(Nan::New(name).ToLocalChecked(), Nan::Undefined());
    ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3639:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3642:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/addon.cpp:42:13: error: no matching member function for call to 'Delete'
    v8obj_->Delete(Nan::New(name).ToLocalChecked());
    ~~~~~~~~^~~~~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3724:37: note: candidate function not viable: requires 2
      arguments, but 1 was provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Delete(Local<Context> context,
                                    ^
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3729:37: note: candidate function not viable: requires 2
      arguments, but 1 was provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Delete(Local<Context> context,
                                    ^
../src/addon.cpp:46:13: error: no matching member function for call to 'Set'
    v8obj_->Set(Nan::New(name).ToLocalChecked(),
    ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3639:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3642:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/addon.cpp:51:13: error: no matching member function for call to 'Set'
    v8obj_->Set(Nan::New(name).ToLocalChecked(), value);
    ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3639:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3642:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/addon.cpp:55:13: error: no matching member function for call to 'Set'
    v8obj_->Set(Nan::New(name).ToLocalChecked(),
    ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3639:37: note: candidate function not viable: requires 3
      arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
[... 90 lines cut ...]
-- Performing Test curl_cv_func_send_test
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const char *, socklen_t, int)
-- Performing Test curl_cv_func_send_test
[ 76%] Linking CXX executable rs-enumerate-devices
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const char *, socklen_t, unsigned int)
[ 76%] Linking CXX shared library librealsense2-gl.dylib
-- Performing Test curl_cv_func_send_test
[ 76%] Built target rs-enumerate-devices
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const char *, unsigned int, int)
-- Performing Test curl_cv_func_send_test
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const char *, unsigned int, unsigned int)
-- Performing Test curl_cv_func_send_test
[ 76%] Built target realsense2-gl
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const void *, int, int)
-- Performing Test curl_cv_func_send_test
-- Performing Test curl_cv_func_send_test - Failed
-- Tested: ssize_t send(int, const void *, int, unsigned int)
-- Performing Test curl_cv_func_send_test
```
Your help is very much appreciated.

MartyG-RealSense, 2021-05-21:
I do not have a reference to a past example of a RealSense Node.js installation on Mac to compare to yours.  A RealSense user in a past case who had problems with the **npm install** instruction did find though that the alternative instruction below worked for them, so you could give that a try:

**npm i --vs-configuration=Release**

https://github.com/IntelRealSense/librealsense/issues/8847#issuecomment-821787303

alifeinbinary, 2021-05-21:
Running that command in `./wrappers/nodejs/` doesn't help. 

It looks to me like that problem is isolated to the `./wrappers/nodejs/src/addon.cpp` file. Whether building from source or installing via npm, it always fails at that spot. 

This is the first error that appears during a build with npm. 

```
** BUILD SUCCEEDED **

> node-librealsense@0.2130.0 install /Users/andrew/projects/eel/node_modules/node-librealsense
> node-gyp rebuild

  CXX(target) Release/obj.target/node_librealsense/src/addon.o
../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
    return v8obj_->Get(Nan::New(name).ToLocalChecked());
           ~~~~~~~~^~~
/Users/andrew/Library/Caches/node-gyp/14.0.0/include/node/v8.h:3686:43: note: candidate function not viable: requires 2
      arguments, but 1 was provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
```

MartyG-RealSense, 2021-05-21:
Thanks very much for your patience.  My research located some guidance about using **node-gyp** with MacOS Catalina that may be worth reading.

https://github.com/nodejs/node-gyp/blob/master/macOS_Catalina.md

alifeinbinary, 2021-05-21:
Wow, that a wild ride. You were right @MartyG-RealSense , it had to do with node-gyp's compatibility issues with Catalina specifically. 

What worked for me. 
Open `~/.node-gyp/<your version>/include/node/common.gypi`, search for `MACOSX_DEPLOYMENT_TARGET` and set it to your current version of Mac OS. My file reported 10.13, even though I'm on 10.15.7, so I corrected it. Next, use `n` or `nvm` to downgrade to node@8.16.1 with `sudo n 8.16.1`. `npm install --save node-librealsense` worked after I did those steps. Thanks so much Marty, you're a livesaver!

closed 2021-05-21 by alifeinbinary

MartyG-RealSense, 2021-05-22:
You are very welcome @alifeinbinary - thanks so much for sharing the details of your MacOS Catalina installation method for Node.js with the RealSense community!   :)