realsenseai/librealsense #9071: Problems with output Point Cloud of the C++API with RGB texture(LIdar Camera L515)

Ziling169BENZHI, 2021-05-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentation](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is your question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open a new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (01.05.05.00) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.44 }                          | 
| Language                            |  {C++/python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
-------------------------------------------------------------------------------------
the reason to start this problem is the ply file exported from RSviewer and save_to_ply.py both came out as the same form 
1elements with 9 properties, but each line only contains 3 elements, so a bit confusing to load the data in python or CPP. 
 -------------------------------------------------------------------------------------
Hi, I have been working on export my ply file with CPP API for about a week; I expect to export the ply file with RGB, meshed point cloud data and normals like in RealsenseViewer.  

I have looked through many helpful issues in the community already ( such as #8656, #6979 and #2832) 
Based on their help, I can export the point cloud with the colourized depth and mesh it like this:

[image]

And I tried to play around with rs2::save_to_ply, but I am not sure how to do that. Can anyone take a look at my code and gives me some help? 
[image]
-------------------------------------------------------------------------------------
1>-----
my idea is to use
 rs2::save_to_ply exporter(fname,pc);  // because save_to_ply can provide set_options then using 
exporter.process(colorized)                // this will help me process the saving procedure

But I can only put colourized depth frames in the exporter.process() method, I am not sure why I cannot put the colour frame inside. 
-------------------------------------------------------------------------------------
2>--------
I also tried to use points.export_to_ply(fname, colour) directly to export the file. However, this will help me export the data with RGB texture, but it didn't provide me options to select mesh or normals; it will be helpful if anyone can provide me with some help to solve this option. (BTW, the ply file saved in this file can only be opened in meshlab, I tried to open it with Python, or Clion turned out to be a total mess.
[image]

Thank you for all the help @MartyG-RealSense @RealSenseSupport

Ziling169BENZHI, 2021-05-27:
also I m confused whether the XYZ coordinates are pixel coordinates or 3D coordinates in the Camera frame

MartyG-RealSense, 2021-07-02:
Hi @Ziling169BENZHI   Did you manage to find a solution for the problem in this case, please?

Ziling169BENZHI, 2021-07-05:
> Hi @Ziling169BENZHI Did you manage to find a solution for the problem in this case, please?

Hi MartyG-RealSense, sorry I still have the same problem with 
rs2:: save_to_ply exporter("",pc);

On the other side, for points.export_to_ply("",color); I figured out the problem is the binary data type reader, do you have any idea, how can I change the format to " format ascii 1.0" ??

Ziling169BENZHI, 2021-07-05:
> Hi @Ziling169BENZHI Did you manage to find a solution for the problem in this case, please?

Also, I am trying to change the minimum range from default to 0, however, I am bit confused about how to implement the sensor control example from SDK2.0 on point cloud example, Can you give me some hints? 

Thanks a lot.

MartyG-RealSense, 2021-07-05:
When using save_to_ply, setting **OPTION_PLY_BINARY** to a value of '0' should result in the ply being exported in ASCII format.

https://github.com/IntelRealSense/librealsense/pull/4906

An example of a C++ script for doing so is here:

https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-650277811

If you wish to shorten minimum range on L515, the link below suggests loading a Visual Preset (json) camera configuration such as **Short Range**.

https://github.com/IntelRealSense/librealsense/issues/7731

I also located a Python script example for setting the L515's minimum distance to '0'.

https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501

Ziling169BENZHI, 2021-07-05:
> When using save_to_ply, setting **OPTION_PLY_BINARY** to a value of '0' should result in the ply being exported in ASCII format.
> 
> #4906
> 
> An example of a C++ script for doing so is here:
> 
> [#6677 (comment)](https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-650277811)
> 
> If you wish to shorten minimum range on L515, the link below suggests loading a Visual Preset (json) camera configuration such as **Short Range**.
> 
> #7731
> 
> I also located a Python script example for setting the L515's minimum distance to '0' using code.
> 
> [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501)

Hi Marty, 

I am sorry I guess I didn't explain my question more clearly. 
So First I tried with 
        rs2::save_to_ply exporter("mesh_face.ply", pc);
        exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 1.f);
        exporter.set_option(rs2::save_to_ply::OPTION_PLY_NORMALS, 1.f);
        exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
        //exporter.invoke(depth);
        exporter.export_to_ply(points, color);
However, this exporter doesn't have exporter.export_to_ply() function: 
[image]

Since I already know there exist two method to export the point cloud to ply file. To change the binary option, I would like to know how to implement this option with method of points.export_to_ply("anotherface.ply", color);

Best 
Ziling

MartyG-RealSense, 2021-07-06:
https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-648079238 states that ASCII (non-binary) ply export is not possible using **export_to_ply** and that save_to_ply should be used instead to do so.

The case https://github.com/IntelRealSense/librealsense/issues/6979 that you have already seen had the same kind of problem as the error that you highlighted above.

https://github.com/IntelRealSense/librealsense/issues/6979#issuecomment-668147554

The RealSense user who created https://github.com/IntelRealSense/librealsense/issues/6979 shared a Python script in a later case that exported the ply with binary = false using the save_to_ply instruction.  The script did not export color though.

https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152

I have mentioned in past cases that discussed export of ply with color that history has shown that exporting a point cloud in Python that includes color information (which the title of this case suggests that you are seeking) is problematic to achieve.

The link below has a Python script that used export_to_ply to export a color ply that lacked normals.  And as mentioned earlier, export_to_ply does not have ASCII export.

https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

I recalled that there was also a case in the link below where a RealSense user created a script for export a color ply with export_to_ply for the purposes of accessing it as ASCII.

https://community.intel.com/t5/Items-with-no-label/Capturing-data-from-realsense-D435-exporting-coloured-depth-to/td-p/692227

```
auto depth = frames.get_depth_frame();
points = pc.calculate(depth);
auto color = frames.get_color_frame();
pc.map_to(color);
 
...
const string st ="filename.ply";
points.export_to_ply(st.c_str(), color);
```

My understanding of that case is that once the RealSense user exported the color ply then they opened the ply file in the open-source **MeshLab** software package and exported the mesh as ASCII.

https://www.meshlab.net/

MartyG-RealSense, 2021-07-12:
Hi @Ziling169BENZHI  Do you require further assistance with this case, please?  Thanks!

Ziling169BENZHI, 2021-07-12:
> Hi @Ziling169BENZHI Do you require further assistance with this case, please? Thanks!

Hi, Marty Thank you very much, unfortunately, I still need some assistance, I am trying to load the Json file into my code, I am using the method from #6739 and #4046. 

Currently, I felt bit confuse about how this works: 
auto advanced_mode_dev = serializable_device(this->p->get_active_profile().get_device());
std::ifstream t(myFile);
std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
if (auto advanced_mode_dev = p->get_active_profile().get_device().as<serializable_device>();)
{
    advanced_mode_dev.load_json(preset_json);
    ....
} 

Besides that, I am wondering how to set the mode to Short Range Set in C++?

MartyG-RealSense, 2021-07-12:
Setting up the loading of a camera configuration json can be a little awkward.  Let's focus on trying to set up the Short Range preset first.  Fortunately the code for doing so is simpler than for custom made presets.  Intel provide a guide for doing so in C++ in the link below.

https://www.intel.com/content/www/us/en/support/articles/000028416/emerging-technologies/intel-realsense-technology.html

If the second of the two methods in that guide is used:

```
auto sensor = profile.get_device().first<rs2::depth_sensor>();
sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET, rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
```

To define the L515 Short Range preset, please try changing this instruction:

**rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY**

To this:

rs2_l500_visual_preset::RS2_L500_VISUAL_PRESET_SHORT_RANGE

Ziling169BENZHI, 2021-07-14:
> Setting up the loading of a camera configuration json can be a little awkward. Let's focus on trying to set up the Short Range preset first. Fortunately the code for doing so is simpler than for [...]

thank you so much, Marty I am able to change the Short-range set. I also have some questions about setting the resolution of the L515, do you have any recommended issue I can use?

MartyG-RealSense, 2021-07-14:
The code for setting the resolution for L515 in C++ will be the same as with other RealSense cameras, except for selecting resolutions that are supported by L515.  You can set up **cfg** configuration instructions just before the pipe start line of the script, and then place 'cfg' in the brackets of the pipe start instruction so that the script knows to use that custom configuration instead of the L515's default stream configuration.

Here's a simple example of such code that is taken from an L515 case for defining depth and color streams:

```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 1024, 768, RS2_FORMAT_Z16, 30); 
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 30); 
rs2::pipeline pipe;
pipe.start(cfg);
```

MartyG-RealSense, 2021-07-26:
Case closed due to no further comments received.

closed 2021-07-26 by MartyG-RealSense

Ziling169BENZHI, 2021-08-12:
> Hi @Ziling169BENZHI Do you require further assistance with this case, please? Thanks!

Sorry I got sick for few weeks. Thank you for all the help.

MartyG-RealSense, 2021-08-12:
No problem at all, @Ziling169BENZHI - you are very welcome!