realsenseai/librealsense #9072: can not open Intel realsense viewer

zhang253, 2021-05-22:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-05-22:
Hi @zhang253   Does this error message appear when you are launching the RealSense Viewer from the Windows desktop, please?

Also, have you installed the full **RealSense SDK** software using the file for the automated SDK installer on the **Releases** web-page?  The file list can be found by finding the word **Assets** at the bottom of the information listing for each SDK version and left-clicking on the arrow beside it to expand open the file list.

https://github.com/IntelRealSense/librealsense/releases

[image]

zhang253, 2021-05-26:
I have installed the full Realsense SDK. The first time when I lauched the RealSense viewer is normal. But after I run a sample code. I can never run the Realsense viewer again.

 

 
 
https://github.com/IntelRealSense/librealsense/releases

zhang253, 2021-05-26:
Align.There is a window showing depth image.

 
Which sample program did you run, please?

MartyG-RealSense, 2021-05-26:
Running the **rs-align** sample should not have had a negative effect on the RealSense Viewer unless rs-align was running when the Viewer was being used.  Then it would likely block streams from being started in the Viewer.  So rs-align should be closed when the Viewer is being used.

MartyG-RealSense, 2021-06-01:
Hi @zhang253  Do you require further assistance with this case, please?  Thanks!

zhang253, 2021-06-01:
Yes, I still can not use InterRealsense viewer.But I can run sample code now.

MartyG-RealSense, 2021-06-01:
Are you able to run the **rs-capture** example program?

closed 2021-06-13 by MartyG-RealSense