realsenseai/librealsense #9073: Error when cloning libjpeg-turbo dependency

buschbapti, 2021-05-22:
When trying to build the library from source, cmake is downloading the `libjpeg-turbo` package. However, it is trying to clone the `master` branch. I have checked on their repository and a few days ago they have changed from `master` to `main`.

MartyG-RealSense, 2021-05-22:
Hi @buschbapti   I believe that you are referring to the current default branch on the **libjpeg-turbo** GitHub.  Is that correct, please?

[image]

I checked the zip-file version of the librealsense source code that is available from the **Assets** file list on the SDK **Releases** page.

https://github.com/IntelRealSense/librealsense/releases

After unzipping the librealsense source code folder, I performed a search from the root folder for the term **turbo** and found in the search results a **CMakeLists.txt** build file that contains a reference to the **master** branch of the libjpeg-turbo GitHub.

[image]

[image]

Conceivably, the branch tag in this CMakeLists.txt file could be edited to **main** before building librealsense using this source code folder.

This block of code suggests though that activation of this code is dependent on the CMake build flag **BUILD_NETWORK_DEVICE** being on / true.  Are you making use of **BUILD_NETWORK_DEVICE** in your source-code build, please?

buschbapti, 2021-05-22:
Yes this is correct. I was actually trying to use the `Dockerfile` from feature request #3849. I guess it might use this build flag indeed.

MartyG-RealSense, 2021-05-22:
A comment on https://github.com/IntelRealSense/librealsense/issues/3849 from April 22 2021 does reference network support in the Docker image.

https://github.com/IntelRealSense/librealsense/issues/3849#issuecomment-824495867

MartyG-RealSense, 2021-05-28:
Hi @buschbapti  Do you have an update about this case that you can provide, please?  Thanks!

buschbapti, 2021-05-28:
I could compile without the flag and #3849 states that they are looking into it. I am mostly interested in the official Docker image when it will be released anyway. Thanks for checking. I think this can be closed.

MartyG-RealSense, 2021-05-28:
Okay, thanks very much @buschbapti - you are very welcome.  I will close the case as suggested.

closed 2021-05-28 by MartyG-RealSense