realsenseai/librealsense #9076: about imu data  time_stamp

Jxpenggf, 2021-05-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.12.13.50) | 
| Operating System & Version |    Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (NULL)                                         | 
| Platform                                 | PC |
| Language                            |  {C++ }                          | 

### Issue Description
I have found a problem using the following code:
`
int main() {

    std::map<int, std::pair<int,double>> counters;
    std::map<int, std::string> stream_names;
    std::mutex mutex;

    auto callback = [&](const rs2::frame &frame)
            {
        std::lock_guard<std::mutex> lock(mutex);
        if (rs2::frameset fs = frame.as<rs2::frameset>())
        {
            // With callbacks, all synchronized stream will arrive in a single frameset
            for (const rs2::frame &f : fs)
            {
                counters[f.get_profile().unique_id()].first++;
                std::cout<<"id:"<<f.get_profile().stream_name()<<"      "<<std::setprecision(20)<<f.get_timestamp()
                <<" accl: "<<counters[5].first<<" "<<counters[5].second<<" "
                <<" gyro: "<<counters[6].first<<" "<<counters[6].second<<"\n";;
            }
        } else
            {
            counters[frame.get_profile().unique_id()].first++;
            counters[frame.get_profile().unique_id()].second=frame.get_timestamp();
        }
    };

    // Declare RealSense pipeline, encapsulating the actual device and sensors.
    const int width = 640;
    const int height = 480;
    int fps = 30;
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, width, height, RS2_FORMAT_BGR8, fps);
    cfg.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16,fps);
    cfg.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
    cfg.enable_stream(RS2_STREAM_INFRARED, 2, width, height, RS2_FORMAT_Y8, fps);
    cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F,250);
    cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F,400);
    rs2::pipeline_profile profiles = pipe.start(cfg,callback);
    // Collect the enabled streams names
    for (auto p : profiles.get_streams())
        stream_names[p.unique_id()] = p.stream_name();
    std::cout << "RealSense callback sample" << std::endl << std::endl;
    while (true) {
        std::this_thread::sleep_for(std::chrono::seconds(1));
        std::lock_guard<std::mutex> lock(mutex);
        for (auto p : counters) {
            std::cout << stream_names[p.first] << "[" << p.first << "]: " << p.second.first << " [frames] || \n";
        }
    }
}
`
I took a screenshot from the output :
id:Depth            1621750853316.3410645  accl: 353 1621750853356.9421387  gyro: 483 1621750853359.9619141
id:Infrared 1      1621750853316.3410645  accl: 353 1621750853356.9421387  gyro: 483 1621750853359.9619141
id:Infrared 2      1621750853316.3410645  accl: 353 1621750853356.9421387  gyro: 483 1621750853359.9619141
id:Color             1621750853326.644043  accl: 353 1621750853356.9421387  gyro: 483 1621750853359.9619141

id:Depth           1621750853349.9599609  accl: 362 1621750853392.4958496  gyro: 496 1621750853392.6181641
id:Infrared 1      1621750853349.9599609 accl: 362 1621750853392.4958496  gyro: 496 1621750853392.6181641
id:Infrared 2      1621750853349.9599609 accl: 362 1621750853392.4958496  gyro: 496 1621750853392.6181641
id:Color             1621750853360.2475586 accl: 362 1621750853392.4958496  gyro: 496 1621750853392.6181641

id:Depth           1621751076423.2236328 accl: 692 1621751076459.6943359  gyro: 1013 1621751076458.723877
id:Infrared 1      1621751076423.2236328 accl: 692 1621751076459.6943359  gyro: 1013 1621751076458.723877
id:Infrared 2      1621751076423.2236328 accl: 692 1621751076459.6943359  gyro: 1013 1621751076458.723877
id:Color             1621751076432.717041  accl: 692 1621751076459.6943359  gyro: 1013 1621751076458.723877
OK, It can be seen from the output that the depth/infra/icolor data is about 25-30ms later than the IMU data. Should the timestamps be consistent with my code?how to explain this?How to make them consistent.

MartyG-RealSense, 2021-05-23:
Hi @Jxpenggf  In our previous discussion about this subject, I highlighted the link https://github.com/IntelRealSense/librealsense/issues/3205 that explains that it is expected that the IMU will be approximately 40 ms ahead of depth data.

Could you try disabling **global time** please to see whether that has a positive effect on your results, as described in the link below?

https://github.com/IntelRealSense/librealsense/issues/6424#issuecomment-631589256

Jxpenggf, 2021-05-23:
> Hi @Jxpenggf Could you try disabling **global time** for the IMU please by setting **RS2_OPTION_GLOBAL_TIME_ENABLED** to '0' to see whether that has a positive effect on your results, as described in the link below?
> 
> [#6424 (comment)](https://github.com/IntelRealSense/librealsense/issues/6424#issuecomment-631589256)

Jxpenggf, 2021-05-23:
@MartyG-RealSense I tried to add the following to the above code
` auto senors = ctx.query_all_sensors();
  for (const auto& sensor:ctx.query_all_sensors()) {
       sensor.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED,0);
}
rs2::pipeline pipe;
`
But it seems that the system is still **RS2_OPTION_GLOBAL_TIME_ENABLED**.Where did I go wrong?

closed 2021-05-23 by Jxpenggf

MartyG-RealSense, 2021-05-23:
Could you try placing it **after** the pipeline has started please?

https://github.com/IntelRealSense/librealsense/issues/7614#issuecomment-713680733

Jxpenggf, 2021-05-23:
@MartyG-RealSense 
tried to use the hardware clock, but the result shows that the depth data is still about 30ms away from the IMU data
`name:Depth   4294228.1390000004321  acc  529   4294266.8679999997839  gyro: 759   4294268.9670000001788 Hardware Clock`

MartyG-RealSense, 2021-05-23:
Can you cofirm please that you placed **RS2_OPTION_GLOBAL_TIME_ENABLED** after the pipe start instruction and that you also set it to '0' (disabled) and not '1' (enabled).