realsenseai/librealsense #9077: Does Cubemos SDK supprort multi camera?

Kismetzc, 2021-05-23:
My camera model is D435i

Kismetzc, 2021-05-23:
Or are there other SDKs that support multiple cameras?  I want to use multiple D435i to achieve motion capture

MartyG-RealSense, 2021-05-23:
Hi @Kismetzc   Intel does not provide technical support for the **Cubemos Skeleton Tracking SDK** as it is a product of the **Cubemos** company and not Intel, so questions about the skeleton tracking SDK should be directed to their website.

https://www.cubemos.com/skeleton-tracking-sdk

In regard to other references for supporting multi-camera motion capture: whilst I am not aware of a fully-formed commercial grade SDK for multiple camera body tracking with RealSense, RealSense certainly can be used for this purpose.

Intel has published an advanced-level seminar on YouTube about using multiple RealSense cameras to train a data-set for body tracking.

https://www.youtube.com/watch?v=VSHDyUXSNqY

You may also be interested in a Python-based multi-camera RealSense motion capture project called **DynaMo**.

https://github.com/anderson-cu-bioastronautics/dynamo_realsense-capture

https://www.researchgate.net/publication/336131252_DynaMo_Dynamic_Body_Shape_and_Motion_Capture_with_Intel_RealSense_Cameras

There is also a project by **VCL** that supports multi-camera capture, though with RealSense D415.

https://github.com/VCL3D/VolumetricCapture

Kismetzc, 2021-05-23:
> Hi @Kismetzc Intel does not provide technical support for the **Cubemos Skeleton Tracking SDK** as it is a product of the **Cubemos** company and not Intel, so questions about the skeleton tracking SDK should be directed to their website.
> 
> https://www.cubemos.com/skeleton-tracking-sdk
> 
> In regard to other references for supporting multi-camera motion capture: whilst I am not aware of a fully-formed commercial grade SDK for multiple camera body tracking with RealSense, RealSense certainly can be used for this purpose.
> 
> Intel has published an advanced-level seminar on YouTube about using multiple RealSense cameras to train a data-set for body tracking.
> 
> https://www.youtube.com/watch?v=VSHDyUXSNqY
> 
> You may also be interested in a Python-based multi-camera RealSense motion capture project called **DynaMo**.
> 
> https://github.com/anderson-cu-bioastronautics/dynamo_realsense-capture
> 
> https://www.researchgate.net/publication/336131252_DynaMo_Dynamic_Body_Shape_and_Motion_Capture_with_Intel_RealSense_Cameras
> 
> There is also a Python project by **VCL** that supports multi-camera capture, though with RealSense D415.
> 
> https://github.com/VCL3D/VolumetricCapture

OK. Thanks for your advice!

Kismetzc, 2021-05-23:
By the way, what about Skeleton Tracking? I only know that NUITRACK SDK supports multiple cameras.

MartyG-RealSense, 2021-05-23:
Nuitrack stated that in a discussion on their forum that multiple camera support was on their development roadmap.  The most recent comment on that discussion in October 2020 suggests that it had not yet been added though.

https://community.nuitrack.com/t/adding-multiple-cameras-to-nuitrack/

Their recent **Nuitrack AI** product provides improved 360 degree tracking but apparently still uses a single camera.

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Nuitrack_AI.md
https://www.youtube.com/watch?v=S8dkf2MEZeM

Kismetzc, 2021-05-23:
> Nuitrack表示，在他们的论坛上的讨论中，他们的发展路线是多摄像头支持。
> 
> https://community.nuitrack.com/t/adding-multiple-cameras-to-nuitrack/
> 
> 但是，与Cubemos一样，英特尔不为Nuitrack提供技术支持。

THX! Do you know which SDKs can do skeleton tracking except for these two SDKs?

MartyG-RealSense, 2021-05-23:
Cubemos and Nuitrack are the two main commercial RealSense-compatible skeleton tracking products.  I do not know of others.

The commercial 3D scanning product **RecFusion Pro** supports multiple RealSense cameras but that may not fit your intended application of motion capture.

https://www.recfusion.net/index.php/en/features

MartyG-RealSense, 2021-05-23:
I conducted further research and found information about **iPiSoft**, which demonstrates motion capture with dual RealSense D455 cameras.

https://www.youtube.com/watch?v=brUmSDhlqc0&t=19s

https://ipisoft.com/

Kismetzc, 2021-05-24:
> I conducted further research and found information about **iPi Motion Capture**, which demonstrates motion capture with dual RealSense D455 cameras in the February 2021 YouTube video link below.
> 
> https://www.youtube.com/watch?v=brUmSDhlqc0&t=74s
> 
> https://ipisoft.com/

Thank you very much for your answers！！！

closed 2021-05-24 by Kismetzc