realsenseai/librealsense #9082: Is it possible to control min and max depth range in c# just like in c++ post-processing exmaple?

DecreasedPower, 2021-05-23:


MartyG-RealSense, 2021-05-24:
Hi @DecreasedPower   Provided below for your reference are images of a C# script shared by a RealSense user for defining minimum and maximum values for the depth image with the **Threshold Filter** post-processing filter.  

https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533

****************

You could alternatively define a min-max 'depth clamp' by creating a **json** camera configuration file (also known as a Visual Preset) and loading it into your program.  Here are examples of the code to set depth clamp min-max in a json:

**"aux-param-depthclampmin": "0",
"aux-param-depthclampmax": "65536",**

The easiest way to generate a json is to set the desired parameters in the RealSense Viewer tool and then use the json export button on the toolbar at the top of the Viewer's options side-panel.

DecreasedPower, 2021-05-24:
> Hi @DecreasedPower Provided below for your reference are images of a C# script shared by a RealSense user for defining minimum and maximum values for the depth image with the **Threshold Filter** post-processing filter. Left-click on the images of the script to view them in full size.
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533
> 
> [image]
> 
> [image]

Wow, thanks a lot. This is exactly what I needed.

MartyG-RealSense, 2021-05-24:
I'm pleased that the script is helpful to you @DecreasedPower 

I note that you re-opened your case after closing it.  Do you require further assistance please?  Thanks!

DecreasedPower, 2021-05-24:
> I'm pleased that the script is helpful to you @DecreasedPower
> 
> I note that you re-opened your case after closing it. Do you require further assistance please? Thanks!

Nah, just my first time doing it)

closed 2021-05-24 by DecreasedPower