realsenseai/librealsense #9085: Python Multiprocesing problems: Frame did not arrive in 5000 when run in seperate process.

brunovollmer, 2021-05-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Ubuntu 20.02  | 
| Kernel Version (Linux Only)    |   5.4.0-56-generic                                  | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.0            | 
| Language                            |  Python                      | 
| Segment			|  Robot                   | 

### Issue Description
Hey everybody,

I have a problem when using my D435 with multiprocessing in Python. I would like to acquire the images from the depth camera in a separate process. Problem is that every time I receive the " Frame didn't arrive within 5000" error. The code in itself works when I run it in the main process. Any ideas what to look out for and where the error could be.

```
class Task(Process, metaclass=ABCMeta):
    def __init__(self, name="base_task", data_storages=[], events={}, priority='low'):
        Process.__init__(self)
        self._name = name
        self._data_storages = data_storages
        self._events = events

        if priority not in ['low', 'high']:
            raise ValueError("unknown process priority")

        self._priority = priority

        # reset restart event for this task
        self._events[f'restart_{self._name}'].clear()

    def run(self):
        while True:

            try:
                self._run()
                print(f"{self._name} finished")
            except Exception as e:
                print(f"Exception: {e}")
                # check if restartable
                if self._check_exception(e):
                    # set restart event
                    self._events[f'restart_{self._name}'].set()

                # kill process
                self._shutdown()
                return

            terminate = False
            for n, e in self._events.items():

                if e.is_set():
                    if n in ['shutdown', 'restart']:
                        terminate = True

            if terminate: break

            if self._priority == 'low':
                print(self._name, "sleep")
                time.sleep(1)

    def _check_exception(self, e):
        # TODO check which exceptions exceptions are restartable and which not
        return True

    def _trigger_event(self, name):
        self._events[name].set()

    def _add_data_to_storage(self, name, data):
        data_storage = self._data_storages[name]
        if data_storage.type == 'queue':
            data_storage.queue.put(data)
        elif data_storage.type == 'shared_mem':
            tmp_shared_mem = shared_memory.SharedMemory(name=data_storage.storage_name)
            shared_mem_data = np.ndarray(data_storage.shape, dtype=data_storage.dtype, buffer=tmp_shared_mem.buf)
            shared_mem_data[:] = data[:]
        else:
            ValueError("unknown data source")

    def _get_data_from_storage(self, name):
        if name not in self._data_storages:
            raise ValueError("unknown data source name")

        data_storage = self._data_storages[name]
        if data_storage.type == 'queue':
            return data_storage.queue.get()
        elif data_storage.type == 'shared_mem':
            tmp_shared_mem = shared_memory.SharedMemory(name=data_storage.storage_name)
            shared_mem_data = np.ndarray(data_storage.shape, dtype=data_storage.dtype, buffer=tmp_shared_mem.buf)
            return np.copy(shared_mem_data)
        else:
            ValueError("unknown data source")

    @abstractmethod
    def _run(self):
        pass

    @abstractmethod
    def _shutdown(self):
        pass

    @property
    def name(self):
        return self._name
```

```
class ImageAcquisitionTask(Task):
    def __init__(self, *args, **kwargs):
        super().__init__(priority='high',
                         *args,
                         **kwargs)

        self._width = 848
        self._height = 480

        self._camera_data = self._find_cameras()
        self._depth_profile = None
        self._pipes = None
        self._init_cameras()
        print(self._camera_data)

    def _find_cameras(self) -> dict:
        """Find all currently connected cameras and return their
        data.

        Returns:
            dict: camera names and their serial number.
        """

        ctx = rs.context()
        self._devices = ctx.devices

        data = {}

        for dev in self._devices:
            name = dev.get_info(rs.camera_info.name)
            serial_number = dev.get_info(rs.camera_info.serial_number)

            data[name] = serial_number

        return data

    def _init_cameras(self):
        pipelines = []

        for name, serial_number in self._camera_data.items():
            # Declare RealSense pipeline, encapsulating the actual device and sensors
            pipe = rs.pipeline()

            # Build config object
            cfg = rs.config()
            cfg.enable_device(serial_number)

            print(serial_number)

            cfg.enable_stream(rs.stream.depth,
                              width=self._width,
                              height=self._height,
                              format=rs.format.z16,
                              framerate=30)
            cfg.enable_stream(rs.stream.color,
                              width=self._width,
                              height=self._height,
                              format=rs.format.bgr8,
                              framerate=30)

            # Start streaming with requested config
            profile = pipe.start(cfg)

            pipelines.append(pipe)

            self._depth_profile = profile
            # Get depth scale to estimate exact depth
            self._depth_scale = (self._depth_profile.get_device().
                             first_depth_sensor().get_depth_scale())

        self._pipes = pipelines

    def _run(self):
        # iterate over both cameras
        for pipe in self._pipes:

            # catch any camera exception
            # -> frame didn't arrive within 5000
            try:
                # wait for next frame of camera
                frames = pipe.wait_for_frames()
            except RuntimeError as e:
                # reset device
                print(f"catched device runtime error: {e}")
                exit()

            # get depth, color and timestamp information
            ts = frames.get_timestamp()
            color_frame = frames.get_color_frame()
            depth_frame = frames.get_depth_frame()

            
            # check if image data is valid
            if color_frame is None or depth_frame is None:
                print("invalid depth or color frame")
                exit()

            # Get color image data
            color = np.asanyarray(color_frame.get_data())
            # Align both images
            align = rs.align(rs.stream.color)
            frames = align.process(frames)

            # Get depth image data
            aligned_depth_frame = frames.get_depth_frame()

            # Scale depth and convert to meters
            aligned_depth_frame = np.asanyarray(aligned_depth_frame.get_data())
            aligned_depth_frame = aligned_depth_frame * self._depth_scale

            depth_frame = np.asanyarray(depth_frame.get_data())
            depth_frame = depth_frame * self._depth_scale

            self._add_data_to_storage('depth_img', depth_frame)

    def _shutdown(self):
        return super()._shutdown()

```

MartyG-RealSense, 2021-05-24:
Hi @brunovollmer   As a starting point in investigating your case, could you check the advice in the link below please to see whether it resolves your problem.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/7365

brunovollmer, 2021-05-24:
Hey @MartyG-RealSense unfortunately this does not help.

MartyG-RealSense, 2021-05-24:
Thanks very much.  The next reference that I would direct you towards is a Python multithreading case that was solved by queuing ndarrays.

https://github.com/IntelRealSense/librealsense/issues/7968#issuecomment-744538550

brunovollmer, 2021-05-24:
The problem is not a buffer overflow but that the frame does not arrive when used in another thread.

brunovollmer, 2021-05-24:
Hey @MartyG-RealSense,

thanks. The problem was that I constructed the pipeline inside the constructor of the process. What solved it was to create it at the beginning of the run() function.

MartyG-RealSense, 2021-05-24:
That's great to know that you found a solution - thanks very much for the update!

closed 2021-05-24 by brunovollmer