realsenseai/librealsense #9089: How to avoid the fake strip in 3D colored information

JimXu1989, 2021-05-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 

| Operating System & Version |   { Linux (Ubuntu 14/16/17)   |                                       | 
| Platform                                 | PC/.  |
| SDK Version                            |  { legacy / 2.<45>.<0> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
The color 3D points information always have some fake strip caused by the resolution miss matched between the 
depth img and the color img, the depth img has a wider sight then the rgb img, so, some of the depth img has to be colored as fake strip, as shown below:

[image]

You can see fake strips in the top and right, how could I avoid it? I want just use the 3D points which are correct colored.

Thanks a lot

MartyG-RealSense, 2021-05-31:
Hi @JimXu1989  Do you require further assistance with this case, please?  Thanks!

JimXu1989, 2021-06-01:
> Hi @JimXu1989 Do you require further assistance with this case, please? Thanks!

I still meet the same problem with realsense viewer

MartyG-RealSense, 2021-06-01:
Let's test whether clearing up some of the large black areas without depth data can improve the image.

Please open up the **Controls** section of the Stereo Module options in the Viewer and find an option called **Depth Units**.  This should be set to '0.001' by default.  Left-click on the pencil icon at the end of the option to input a manual value with the keyboard, and set it to **0.0001**.

[image]

This should hopefully cause the amount of empty depth on the image to reduce significantly, as shown in the images below where the upper image represents the default depth unit value of '0.001' and the lower image represents the same scene at '0.0001'.

[image]

[image]

JimXu1989, 2021-06-02:
> Let's test whether clearing up some of the large black areas without depth data can improve the image.
> 
> Please open up the **Controls** section of the Stereo Module options in the Viewer and find an option called **Depth Units**. This should be set to '0.001' by default. Left-click on the pencil icon at the end of the option to input a manual value with the keyboard, and set it to **0.0001**.
> 
> [image]
> 
> This should hopefully cause the amount of empty depth on the image to reduce significantly, as shown in the images below where the upper image represents the default depth unit value of '0.001' and the lower image represents the same scene at '0.0001'.
> 
> **0.001**
> 
> [image]
> 
> **0.0001**
> 
> [image]

Great thanks for your reply, while, the strips may still be there when save the 3d sense to a .ply file

MartyG-RealSense, 2021-06-02:
Do the lines still occur if you reset the Viewer (which is different from resetting the camera hardware) using the instructions in the link below?

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914/comments/1500000416522

closed 2021-06-03 by JimXu1989