realsenseai/librealsense #9092: How can I get the pixel coordinate [0,0] of 3D View

ellestpd, 2021-05-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10| 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.43.0 }                          | 
| Language                            |  {python}                          | 

[image]
[image]

I want to know the pixel of red dot in the first picture to use the rs2_deprojection_pixel_to_point for the real world coordinate point by input the pixel of RGB image or BW image(Left Monochrome)

What I know
1. depth image size and center of camera by read the intrinsic of depth frame (848x480)
2. Full color image size is 1920x1080
3. I have test moving the pixel from the center of camera and measure in real world and that's correct.
4. I test all of this by editing the librealsense\wrappers\python\examples\opencv_pointcloud_viewer.py
5. When I use rs2_deprojection_pixel_to_point( intrinsic , [0,0] ,d ) the program deproject to red point in second picture.

I can't find the way to finish my project if anyone have an idea please share.
Thankyou for the future responds.

MartyG-RealSense, 2021-05-25:
Hi @ellestpd  My understanding is that your code is based on the Python example program **opencv_pointcloud_viewer.py** (as you mention above) and the program is being used to generate a textured point cloud that maps the RGB image to the depth points, similar to the capability to do so in the 3D point cloud mode in the RealSense Viewer tool.  Is that correct, please?

On the D435 / D435i camera models, the size of the depth sensor's field of view (FOV) is larger than the color sensor's FOV, as illustrated by your blue and yellow box image.  The color sensor therefore cannot see color detail in the real-world scene that is outside of its FOV (such as the location of the red dot).  Your diagram indicates that the location is within the field of view of the depth sensor though. 

Possibilities that I can think of for dealing with this situation include:

1.  Align **color to depth** so that color is mapped to the depth image instead of depth to color alignment (where the color FOV size, which may be smaller on D435 / D435i, is used).

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-3-4-stream-alignment

The image below, captured with a D435i in the **rs-align** eample program, shows alignment to color in the upper image (with the cut-off area around the border highlighted) and alignment to depth in the lower image (where the image is resized to the depth FOV size).

[image]

2.  Use a RealSense D455 camera model if you are not doing so already, where the border area on the depth to color alignment is smaller, as illustrated below (D435i upper, D455 lower).

[image]

3.  Widen the overall FOV by arranging two cameras horizontally and stitching their point clouds together into a single large point cloud image.

ellestpd, 2021-05-27:
thankyou for the reply.
I will try this and understand more how it works.
I'll be back for the result.

ellestpd, 2021-05-27:
@MartyG-RealSense
 I've tried the first solution and its works with resolution 640*360 of RGB image as image below.
as you can see the purple line I point to is (0,0) of image and this is correct.
[image]

but if I want to increase the resolution it looks so strange. 
the image rotation and position are shift to the left side if the screen.
How can I solve this problem that make the first image to better resolution

[image]

this is how I change the resolution. I'm not sure that is this the correct way to do.
[image]

MartyG-RealSense, 2021-05-27:
What are the results if you set color to **1280x720** too so that both streams have the same resolution, please?

ellestpd, 2021-05-27:
[image]

This result is from 1280*720 as shown on the window information bar.

MartyG-RealSense, 2021-05-27:
And how about at **848x480** which is in the middle of the range between 640x360 and 1280x720

ellestpd, 2021-05-27:
[image]

or maybe the window was set to resolution 640*360 but i cant find where in the code.

MartyG-RealSense, 2021-05-27:
Increasing the resolution does seem to progressively shift the image sidewards.

If we are stretching the color size to fit to the depth size, how would 640x360 color and 1280x720 depth appear?

ellestpd, 2021-05-27:
its OK now.
I solved my problem by edited the scale value at the top of program to False.
and this is my result. Thankyou so much for many information.

[image]
[image]

MartyG-RealSense, 2021-05-27:
Excellent news @ellestpd that you were successful - thanks very much for the update and for sharing the method that worked for you!   :)

ellestpd, 2021-05-27:
@MartyG-RealSense I wish that my problem will help others in the future thanks to you too.

closed 2021-05-27 by ellestpd