realsenseai/librealsense #9093: Cannot access L515 color camera in Unity

jamestkpoon, 2021-05-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-50-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.45.0 }                          | 
| Language                            |  unity 2020.3.3f1                         | 
| Segment			|  Robot/Smartphone/VR/AR/others                          | 

### Issue Description
I seem unable to run the color sensor in Unity. The Scene that is most relevant to my use case is TexturesDepthAndColor but I can only get it to run by removing the Color profile in the RsDevice prefab, otherwise I get the following error:

ExternalException: rs2_pipeline_start_with_config(pipe:0x55e73e2625f0, config:0x55e737d9b150)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <62652b3ba0a443e98256a4bb99fef2fd>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <62652b3ba0a443e98256a4bb99fef2fd>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

The RealSense Viewer lets me visualize color point clouds so it's not a hardware issue, and I have also used the color camera in Python. Any advice would be welcome. Thanks!

RealSenseSupport, 2021-06-21:
@jamestkpoon 

Were you able to get things working?  Can you let us know with the latest 2.47 SDK and latest FW for L515 does your setup work?

RealSenseCustomerSupport, 2021-07-06:
----------------------------------------------

@jamestkpoon, did you have a chance to try the latest SDK and firmware to see if your issue has been resolved?

jamestkpoon, 2021-07-09:
Sorry for the late reply, back from vacation. Can confirm latest SDK/FW allows color camera access in Unity for me, thank you!

Closing.

closed 2021-07-09 by jamestkpoon