realsenseai/librealsense #9094: 3d reconstruction to 2d array

skakavyponik, 2021-05-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
<My problem is that with : 
        depth_image_out = np.asanyarray(aligned_depth_frame.get_data())
        color_image_out = np.asanyarray(color_frame.get_data())
Ive got aligned frames so on (0,0) is some color and some depth 
but I cannot find solution if I want reprojection from smth like pointcloud to 2d picture
To understand better (sry for bad english) I want to recreate this two arrays (depth and color) to another two arrays but with correction with depth of objects so if smth is closer to camera its bigger so i want it to shrink (like when you are looking at pointcloud)
so lets say I will tell I want array  and filter everything for example more than 2 meters away and I will calculate it into array same size but with shrinked objects depends on depth so it will be accurate to real x,y size 

Thanks >

skakavyponik, 2021-05-25:
lets say I want both depth+color data but for every pixel I ve got values in this case 
But what I really want is smth like view from camera with transformation to "real" system so the x,y will be like at layer with some depth (for example 2 meters)
so if there is object lets say 10x10 pixels in depth 1 meter after transformation it will be 5x5 on 2 meters depths and shadow
My problem now is : 
Camera is checking if there is smth on box2 but there is some shadow from box1 and it looks like its box1 on box2 
but for me it should be ok because in reality on x array it is on good position 
so what I want is to transform every pixel to coordinate system of floor layer depends on depth 
I can do it for every pixel in for cycle but it will take a long time 
[image]
Honestly I am afraid there is no solution for this

MartyG-RealSense, 2021-05-25:
If the main problem is dealing with the possibility of shadows generating incorrect depth readings, a simple solution may be to enable the camera's in-built **projector**.  When enabled, it projects a light-source that is most intense at short range from the camera and also casts a semi-random pattern of dots onto the surface of objects in a scene,  The camera can use these dots that are projected onto surfaces as a 'texture source' to analyze fo depth information.  

Dot-pattern analysis is particularly useful on surfaces that have low texture or no texture that would otherwise make them difficult to read depth from, such as doors, desks, chairs and walls.

The link below provides Python scripting for setting the status of the IR emitter with **depth_sensor.get_option(rs.option.emitter_enabled)**

Set it to '1' to enable it instead of the '0' (disable) that the script uses.

community.intel.com/t5/Items-with-no-label/How-to-enable-disable-emitter-through-python-wrapper/m-p/547902#M8384

The case in the link below, where an overhead camera observing boxes on a conveyor also has to deal with shadow, may also be a useful reference.

https://github.com/IntelRealSense/librealsense/issues/7021

skakavyponik, 2021-05-25:
Honestly I rather want to have shadow than picture of box 1 on box 2 
My problem is calculation of box to x,y coordinates of floor so every box will have the same area at x,y coor 
so closer to camera the more I need to shrink area of box 
So for every point I need to calculate his x,y coor on the floor depends on depth and x,y coor at that depth 
Thats why I mentioned pointcloud because in pointcloud corner of box1 has the same x,y coor as corner of box under box 1 (if it was visible) 
But I need it in 2D matrix 
I am ok with that the box 2 will have shadow on it

MartyG-RealSense, 2021-05-25:
It sounds as though you want to take a 3D scene (a row of boxes of different heights) and flatten the image down so that it is like a sheet of paper laid on the floor (the 'floor layer coordinates') with a picture of boxes on it, so that all the boxes on the flat 2D picture seem to be at the same height from the floor. Is that correct please?

If you needed a point cloud that was completely flat, conceivably you could program a check of the z value of all points that if z does not equal '1' then z = 1.  In that way, the entire point cloud should have a height of 1, like a printed-out image on a sheet of paper.

******************

I wonder though if you could alternatively just drop the depth image and the pointcloud and only have an overhead RGB image.  Because if you are only interested in 2D x and y pixel coordinates, if the camera is facing straight down from overhead then a coordinate on a box that is 1 meter lower should still be at the same xy coordinate as a taller box in the same position, I would think.

skakavyponik, 2021-05-27:
Thanks for your help
I think that my solution is just take all coordinates in "for cycle" and choose ones that I need from points in pointcloud 
I wanted to avoid this "for cycle" because of CPU power 
So I will need to lower resolution

closed 2021-05-27 by skakavyponik

MartyG-RealSense, 2021-05-27:
Thanks very  much @skakavyponik for the update - good luck with your project!